immortalwrt/immortalwrt #981: 使用PVE创建KVM安装，无法进入管理后台

Binsect, 2023-06-17:
### Describe the bug

[image]

使用PVE创建KVM运行，可以正常引导系统，但是会卡在图中位置，并且无法进入管理后台

### ImmortalWrt version

r26886-1bd92542b9

### ImmortalWrt target/subtarget

x86/64

### Device

QEMU Standard PC (i440FX + PIIX, 1996)

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-06-17:
> kvm: no supported hardware

那恐怕需要检查你自己的 pve/bios 设定了。

kvm-intel 是给 openwrt 本身添加 kvm 虚拟支持，你要在 openwrt 内嵌套虚拟机？

Binsect, 2023-06-18:
因为使用默认设置编译immotalwrt无法进入管理后台，尝试性的加入了kvm-intel模块。

使用同样的设置编译lede和原版openwrt可以进入管理后台，还有什么选项可以尝试么

Binsect, 2023-06-18:
补充下，使用21.02分支编译后，可以正常进入后台。
使用master分支编译的进不去

Binsect, 2023-06-18:
使用21.02比较稳定了，不折腾23版本了。

closed 2023-06-18 by Binsect