immortalwrt/immortalwrt #1002: 缺少来自运营商的 IPv6 DNS 配置

wordlesswind, 2023-07-02:
### Describe the bug

运营商通过 IPv6 RA 下发 IPv6 配置，其中包含 IPv6 DNS。
但是 ImmortalWrt 没有使用它。

### ImmortalWrt version

r26701-8a9a8711b2

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

Lunzn FastRhino R68S

### Image kind

Official downloaded image

### Steps to reproduce

在 PPP 链路上启用 IPv6 协商

### Actual behaviour

成功获取 IPv6 配置，包括 DNS：

`ifstatus wan_6`

```
...
"dns-server": [
                        "240e:1f:1::1"
                ],
...
```

但是没有使用它

`/tmp/resolv.conf.auto`

```
# Interface wan
nameserver 202.96.128.86
nameserver 202.96.134.33
# Interface wan_6
```

### Expected behaviour

`/tmp/resolv.conf.auto`

```
# Interface wan
nameserver 202.96.128.86
nameserver 202.96.134.33
# Interface wan_6
nameserver 240e:1f:1::1
```

### Additional info

https://forum.openwrt.org/t/missing-dns-for-dhcpv6-pd/164440

这应该是来自上游的问题，但上游不接受来自分支的 BUG 报告。

如果我在接口中配置自定义 IPv6 DNS，其可以正常配置，但是不会被划分至 `wan_6`；在`状态`——`概览`中，也不会显示 IPv6 DNS。

`/tmp/resolv.conf.auto`

```
# Interface wan
nameserver 2402:4e00::
nameserver 2400:3200::1
nameserver 119.29.29.29
nameserver 223.5.5.5
# Interface wan_6
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-07-03:
> 运营商通过 IPv6 RA 下发 IPv6 配置，其中包含 IPv6 DNS。
但是 ImmortalWrt 没有使用它。

我暂时没有办法复现你的问题，在我的 resolv 文件中正常包含 v6 dns。

> 如果我在接口中配置自定义 IPv6 DNS，其可以正常配置，但是不会被划分至 wan_6

这个是预期行为，因为是在“wan”配置的。

wordlesswind, 2023-07-03:
> 这个是预期行为，因为是在“wan”配置的。

但是这导致 IPv6 DNS 不会显示在概览中。

1715173329, 2023-07-03:
这也是预期行为。

wordlesswind, 2023-07-03:
> 在我的 resolv 文件中正常包含 v6 dns。

重新安装后确实在 `/tmp/resolv.conf.auto` 出现了 IPv6 DNS。

closed 2023-07-03 by wordlesswind