immortalwrt/immortalwrt #1003: 360T7 packet loss on wireless with WED=on

czkwg8, 2023-07-02:
### Describe the bug

如题所指，开启warp也就是wed只有就可以观察到明显的丢包现象
这个问题似乎在官方固件以及immortalwrt-798x中也是一样的
无法确定是mt7981的问题还是路由批次问题（v2ex上也有一个帖子提到类似的情况）

测试方法是使用iperf3 udp测速（因为wed只能加速tcp udp包）
是否有更加新的wed驱动或者相关可能影响的patch可以尝试？

### ImmortalWrt version

snapshot

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

360t7

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

padavanonly, 2023-07-03:
iperf3的参数和对应的测试结果拿出来啊

czkwg8, 2023-07-03:
说法的确是这么回事，理论上都不会过wed，但是开了就是会丢包，原因不明
iperf3 -s -p 8080
iperf3 -c 192.168.1.1 -p 8080 -u -t 30
一般来说30秒测试时间以内都可见丢包
确实无线网络丢个包有什么奇怪的
但是关闭wed之后测试600秒也没有丢包
可能这样测试不是十分科学，主要是换了这个路由之后游戏网络中断卡住和掉线的情况是比较明显
但毕竟过程中变量太多，前段时间也是手动开了wed使用才得到这样的结果
我关掉wed用几天看看吧

padavanonly, 2023-07-03:
> 说法的确是这么回事，理论上都不会过wed，但是开了就是会丢包，原因不明
> iperf3 -s -p 8080
> iperf3 -c 192.168.1.1 -p 8080 -u -t 30
> 一般来说30秒测试时间以内都可见丢包
> 确实无线网络丢个包有什么奇怪的
> 但是关闭wed之后测试600秒也没有丢包
> 可能这样测试不是十分科学，主要是换了这个路由之后游戏网络中断卡住和掉线的情况是比较明显
> 但毕竟过程中变量太多，前段时间也是手动开了wed使用才得到这样的结果
> 我关掉wed用几天看看吧

你这是路由器本身开iperf3，到路由器cpu的tcp udp无论怎样都不会过wed。无线没ack机制，引起丢包的可能性太多了。

closed 2023-07-10 by 1715173329

czkwg8, 2023-09-24:
几个月后回头观察基本可以确定是intel无线网卡驱动的问题
更新到比较新的版本之后就没这现象了