immortalwrt/immortalwrt #1024: 无法通过Zerotier的IP地址访问路由器管理页面

tarvfx, 2023-08-02:
### Describe the bug

无法通过Zerotier的IP地址访问路由器管理页面，页面显示`ERR_CONNECTION_REFUSED`；`ping`显示`无法连接到端口`。只有在路由器本机通过`curl 10.0.0.161`可以访问。

Zerotier 连接状态：

[image]
[image]

Zerotier局域网其他设备访问情况：

[image]
[image]

路由器本机通过curl可以访问

[image]

### ImmortalWrt version

r26809-8fb22b51b6

### ImmortalWrt target/subtarget

x86/64

### Device

VMware, Inc. VMware Virtual Platform

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

Saxon-Sun, 2023-08-02:
防火墙—常规设置——入站数据——改成接受

tarvfx, 2023-08-02:
> 防火墙—常规设置——入站数据——改成接受

感谢您关注这个问题，按照您的设置后现在可以访问了。

closed 2023-08-02 by tarvfx