immortalwrt/immortalwrt #1028: 有关仓库简介中的“中国大陆(mainland China)”英文翻译错误。

cwuom, 2023-08-04:
"_An opensource OpenWrt variant for mainland China users._"中的"mainland China"翻译错误，因改成"**Chinese mainland**"或"**the mainland of China**"又或是"**China's mainland**"。希望可以更正此问题，在国际传播中不应出现这种错误用法

xlighting2017, 2023-08-04:
[image]

https://english.www.gov.cn/services/202304/18/content_WS5e04749bc6d0554996894a1a.html
[image]

http://au.china-embassy.gov.cn/eng/lsfw_12/vc/201305/t20130509_1002348.htm
[image]

https://en.safea.gov.cn/photonews/200609/t20060919_36160.htm
[image]

http://en.moe.gov.cn/news/press_releases/201909/t20190930_401810.html
[image]

Fangliding, 2023-08-04:
大哥第一天上网吗 中国大陆基本都是怎么翻译的

MarksonHon, 2023-08-04:
英文写地名就是从小地名往大地名写，mainland China 符合英语语境

1715173329, 2023-08-04:
以上。

closed 2023-08-04 by 1715173329

cherinyy, 2023-08-08:
> 在国际传播中不应出现这种错误用法

对于普通人而言，只要知道是中国大陆（内地）就行，纠结“Mainland China”还是“the Chinese mainland”实则是陷入了政治陷阱之中。我自己会采用“the Chinese mainland”，但我不会去改变别人的用法。

就好比这几天所谓的“究竟是账号还是帐号”：

> 最新一篇为2022年4月30日。中国社会科学院语言研究所新型辞书编辑室主任储泽祥表示，《现代汉语词典》第7版只收“账号”，不收“帐号”。《现代汉语词典》第7版和《新华字典》第12版，都在“帐”字条目下注明表示账簿、账目时，“旧同‘账’”。也就是说，“帐号”是旧时用法，现在应该用“账号”。

普通人知道是什么就行了，我为什么非要跟随你《现代汉语词典》和《新华字典》的说明？

然后搞得所有公司开始更正自己的文本，真的无语……另一种形式的文字狱罢了。