immortalwrt/immortalwrt #1029: High CPU usage when speedtest.

butinhi, 2023-08-06:
Pls help me how to fix this problem. I test speed or download something with very high CPU usage

[image]

butinhi, 2023-08-08:
> does this problem occur on a wired or wireless network? is hardware offload enabled in `network/firewall`?

I turned on it but not effect

cherinyy, 2023-08-08:
It seems that you are using the SNAPSHOT version of the firmware, I'm not sure about this. Can you try the [23.05-SNAPSHOT](https://downloads.immortalwrt.org/releases/23.05-SNAPSHOT/targets/mediatek/mt7622/immortalwrt-23.05-snapshot-r26857-f853776af6-mediatek-mt7622-xiaomi_redmi-router-ax6s-squashfs-factory.bin)  version please?

cherinyy, 2023-08-08:
> You used this version and not got problem with CPU usage?

I don't have an AX6S so my situation may not apply to you. But I have no problem with my JCG Q30 PRO, which is on the MT7981 platform.

butinhi, 2023-08-08:
> > You used this version and not got problem with CPU usage?
> 
> I don't have an AX6S so my situation may not apply to you. But I have no problem with my JCG Q30 PRO, which is on the MT7981 platform.
Maybe problem with mt7622 (ax6s)

1715173329, 2023-08-09:
I guess this is a upstream issue. Anyway mt7622 is a deprecated platform, just sell it and buy mt7981.

This branch has the best mt7622 support btw.
https://github.com/padavanonly/immortalwrtARM/tree/mt7622

butinhi, 2023-08-09:
> I guess this is a upstream issue. Anyway mt7622 is a deprecated platform, just sell it and buy mt7981.
> 
> This branch has the best mt7622 support btw. https://github.com/padavanonly/immortalwrtARM/tree/mt7622

I am using from this source. It's good for mt7622 but it's old and not updated

olimpiumarius, 2023-08-12:
@This problem is from the speed test software that is from the Microsoft Store, the problem must be sent to this program, whatever router you use does this, I use wrt1900acs v2 with ddwr and openwrt does the same!

butinhi, 2023-08-13:
> @this problem is from the speed test software that is from the Microsoft Store, the problem must be sent to this program, whatever router you use does this, I use wrt1900acs v2 with ddwrt and openwrt does the same!

Same problem with download something from internet

cherinyy, 2023-08-14:
This is the hardware offload not taking effect, you need to find available firmware.

butinhi, 2023-08-14:
> This is the hardware offload not taking effect, you need to find available firmware.

I changed to lede openwrt which has turboacc with MTK WED WO offloading option. It solved my problem. I hope immortalwrt will have it on future
[image]

cherinyy, 2023-08-15:
> I hope immortalwrt will have it on future

You may need to [report this issue upstream](https://github.com/openwrt/openwrt/issues/new?assignees=&labels=bug&projects=&template=bug-report.yml), we don't have the same equipment to test with.

shown19, 2023-08-17:
Hi guys, I know this is not related to the main issue, but just wondering if is it possible to bridge the WIFI to a VLAN instead of a default br-lan? Let see I have created a vlan22, and I want it to be used by the WIFI. Is it possible? I tried on my part but it's not working. 

`
config device
option name 'br-lan'
option type 'bridge'
list ports 'lan1'
list ports 'lan2'
list ports 'lan3'
list ports 'lan4'
list ports 'ra0' <-------- Here's my 2.4ghz WIFI
list ports 'rai0'

config interface 'lan'
option proto 'static'
option netmask '255.255.255.0'
option ip6assign '60'
option device 'br-lan.1'
option ipaddr '192.168.100.1'

config bridge-vlan
option device 'br-lan'
option vlan '1'
list ports 'lan1'
list ports 'lan4'

config bridge-vlan
option device 'br-lan'
option vlan '22'
list ports 'lan1:t'
list ports 'lan2'
list ports 'lan3'
list ports 'ra0' <--------  in hope it will work but it didn't
`

commit 6da77bbf (2024-01-06): mediatek: enable wed offloading for mt7622 by default

Closes: #1029
commit 1e499f35 (2024-01-06): mediatek: enable wed offloading for mt7622 by default

Closes: #1029

(cherry picked from commit 6da77bbfcf51ce73f0562453a52e1a859bd0c44c)
closed 2024-01-06 by 1715173329