immortalwrt/immortalwrt #1030: UPnP does not work

tarvfx, 2023-08-08:
### Describe the bug

路由器部署在家庭网络中主路由位置，上网方式为`pppoe`（无公网`IPv4`，有公网`IPv6`），UPnP无法正常工作。

> 以前使用LEDE开启UPnp服务后，在UPnP列表中可以看到Zerotier端口被映射。

UPnP设置截图：

[image]

和UPnP相关的系统日志
```log
Tue Aug  8 11:54:07 2023 cron.err crond[20973]: crond (busybox 1.36.1) started, log level 5
Tue Aug  8 11:57:22 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 11:57:22 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 11:57:22 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 11:57:22 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 11:57:22 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 11:57:22 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 11:57:22 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 11:57:22 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 11:57:22 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 11:57:22 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 11:57:22 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 11:57:22 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 11:57:22 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 11:57:22 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:02:27 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:02:27 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:02:27 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:02:27 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:02:27 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:02:27 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:02:27 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:02:27 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:02:27 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:02:27 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:02:27 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:02:27 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:02:27 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:02:27 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:07:32 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:07:32 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:07:32 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:07:32 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:07:32 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:07:32 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:07:32 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:07:32 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:07:32 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:07:32 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:07:32 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:07:32 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:07:32 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
Tue Aug  8 12:07:32 2023 daemon.notice miniupnpd[6531]: private/reserved address 100.69.134.82 is not suitable for external IP
```

### ImmortalWrt version

r26836-a59b9d368e

### ImmortalWrt target/subtarget

x86/64

### Device

O.E.M O.E.M（CPU：N3160  网卡：i210）

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

cherinyy, 2023-08-08:
没有互联网 IPv4 地址自然无法工作，需要开启 STUN，服务器可尝试使用 `stun.qq.com`，不一定有效。

tarvfx, 2023-08-08:
你好 @cherinyy 

是因为无公网IPv4的原因吗？之前用LEDE也是同样的网络环境但却可以看到UPnP列表中有Zerotier的端口被映射哦。

1715173329, 2023-08-08:
immortalwrt 23.05+ 必须要有公网才可使用 upnp。不会再处理了，要 upnp 可以降级 21.02 或使用 lede。

closed 2023-08-08 by tarvfx

tarvfx, 2023-08-09:
> 对，在`MiniUPnP 设置`——`高级设置`里启用`使用 STUN`有可能有用。

按您的设置，开启STUN后最终UPnP映射还是不成功。我在调试日志中看到上游NAT为端口限制型或对称型，并且上游不支持转发，所以即便路由器开启了UPnP其实也没有任何意义了是吧？

cherinyy, 2023-08-09:
> 上游NAT为端口限制型或对称型，并且上游不支持转发

那确实没有意义了，无法真正打通内外网。

igmkgu, 2025-05-12:
请问现在的24.10可以开箱即食upnp了吗，还是说依然要求公网ipv4？