immortalwrt/immortalwrt #1034: PVE虚机机在概览页的端口状态不显示虚拟网卡

shenlijun, 2023-08-11:
### Describe the bug

直通进来的实体网卡没有这个问题。

### ImmortalWrt version

r26858-63caa6875f

### ImmortalWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Steps to reproduce

PVE8安装wrt虚拟机,给它一个虚拟网卡就可以复现，我有4台服务器都是同样的问题

### Expected behaviour

概览页可以显示所有网口，不管是否是虚拟的

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

closed 2023-08-11 by shenlijun

1715173329, 2023-08-11:
虚拟网卡缺少必要的数据，无法显示。

shenlijun, 2023-08-11:
> 虚拟网卡缺少必要的数据，无法显示。

是新版luci的问题么？刚才又试了下LEAN的固件，是可以显示的

1715173329, 2023-08-11:
lean 的 autocore 与我们的实现方式不同，如需显示可以继续用 lean 的固件。

shenlijun, 2023-08-12:
> lean 的 autocore 与我们的实现方式不同，如需显示可以继续用 lean 的固件。

我看看月底以后有空，试试参考下那边的代码，能不能搞个PR

1715173329, 2023-08-12:
不必了，这部分的实现不会改动。