immortalwrt/immortalwrt #1037: ddns "stop ddns"clicks are unresponsive

science2468, 2023-08-13:
### Describe the bug

"stop ddns", "restart ddns", "Update DDns Services List" all click unresponsively

### ImmortalWrt version

23.05.0-rc3-r26804-a3639e76d6

### ImmortalWrt target/subtarget

x86/64

### Device

ASUSTeK COMPUTER INC. F1A55-M LX3 PLUS R2.0

### Image kind

Official downloaded image

### Additional info

[image]

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-08-13:
You don't even enable them, why do you think they will work?
If you don't know how to use it, just install ddns-go instead.

closed 2023-08-13 by 1715173329

science2468, 2023-08-14:
> You don't even enable them, why do you think they will work? If you don't know how to use it, just install ddns-go instead.

ddns is working, but clicking "stop ddns", "restart ddns", "Update DDns Services List" does not respond. On the system-startup can start and shut down ddns.

1715173329, 2023-08-14:
I can only see log says "Service section disabled".
Paste your full ddns configuration here.

1715173329, 2023-08-14:
This is not a bug and not related to core repo. Moved to discussion.