immortalwrt/immortalwrt #1050: Broken Firmware builder image mvebu/cortexa9

Hayo-Tee, 2023-08-27:
### Describe the bug

I'm trying to create firmware through a firmware builder with packages:

`block-mount btrfs-progs fdisk fstools irqbalance kmod-fs-btrfs kmod-fs-cifs kmod-fs-exportfs kmod-fs-f2fs kmod-fs-isofs kmod-fs-nfs kmod-fs-nfs-common kmod-fs-nfs-common-rpcsec kmod-fs-nfs-v4 kmod-fs-nfsd kmod-fs-ntfs kmod-usb-core kmod-usb-ehci kmod-usb-ledtrig-usbport kmod-usb-net kmod-usb-net-cdc-ether kmod-usb-net-cdc-mbim kmod-usb-net-cdc-ncm kmod-usb-net-huawei-cdc-ncm kmod-usb-net-qmi-wwan kmod-usb-net-rndis kmod-usb-printer kmod-usb-serial 
kmod-usb-serial-option kmod-usb-serial-wwan kmod-usb-storage kmod-usb-uhci kmod-usb-wdm kmod-usb2 kmod-usb3 luci luci-base luci-mod-admin-full luci-theme-bootstrap luci-app-firewall luci-app-advanced-reboot luci-app-attendedsysupgrade luci-app-qos luci-app-upnp luci-app-wifischedule luci-app-wol luci-compat luci-proto-3g luci-proto-hnet luci-proto-ipv6 luci-proto-modemmanager luci-proto-ncm luci-proto-ppp luci-proto-qmi luci-proto-relay mc modemmanager nano ntfs-3g htop mount-utils usbutils kmod-usb-storage-uas luci-app-zerotier`

But after the flashing the router, the router goes to reboot several times, and returns to the alternative partition.
I could'nt get logs, I lost my TTL-UART converter.

Device is LINKSYS WRT1900ACS v1

attached the old firmware from March, and the broken one

[OLD_FROM_MARCH.zip](https://github.com/immortalwrt/immortalwrt/files/12447380/OLD_FROM_MARCH.zip)
[BROKEN.zip](https://github.com/immortalwrt/immortalwrt/files/12447381/BROKEN.zip)

### ImmortalWrt version

21.02.06 and ImmortalWrt SNAPSHOT r25487-8df50f3b41

### ImmortalWrt target/subtarget

mvebu/cortexa9

### Device

LINKSYS WRT1900ACS

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-08-27:
__Troubleshooting any problem without the error log is like driving with your eyes closed.__
-- _Apache Getting Started_

Hayo-Tee, 2023-08-27:
> **Troubleshooting any problem without the error log is like driving with your eyes closed.** -- _Apache Getting Started_

I know, But At the moment I can't get logs. Can only wait delivery new one from Ali.

Hayo-Tee, 2023-10-08:
Hello. I don't see any bug in new builds.

closed 2023-10-15 by 1715173329