immortalwrt/immortalwrt #1051: Hinink H6xK系列的一个combo phy现在默认定义改为sata了

Beaverfffan, 2023-08-30:
### Describe the bug

[image]
Hinink H6xK系列的一个combo phy现在产品默认定义为sata了.usb host0大概需要关闭？

### ImmortalWrt version

master

### ImmortalWrt target/subtarget

latest

### Device

Hinlink H69K

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

Beaverfffan, 2023-08-30:
[kernellog.txt](https://github.com/immortalwrt/immortalwrt/files/12474817/kernellog.txt)
[systemlog.txt](https://github.com/immortalwrt/immortalwrt/files/12474819/systemlog.txt)

1715173329, 2023-09-19:
本项目即将移除所有 hinlink 支持，使用原厂优秀固件即可。