immortalwrt/immortalwrt #1052: Luci界面无法使用上传文件相关功能

OldCoding, 2023-08-30:
### Describe the bug

使用master分支源码编译固件，包含了文件传输、晶晨宝盒、某clash等插件，但是在上传文件的时候提示没有上传证书，并且无法传输完成，上传的文件也不完整，只有几k大小
[image]
[image]
尝试切换到老分支，比如21.02编译的固件就可以正常上传

### ImmortalWrt version

r27421-6085f6aa7c

### ImmortalWrt target/subtarget

armsr/armv8

### Device

Phicomm N1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-08-30:
1. 不支持的设备
2. 非本项目开发的插件
3. “文件传输”已无意义，软件包页面直接支持上传 ipk

OldCoding, 2023-08-30:
> 新 LuCI 已不支持传统 lua 插件的文件上传功能，重写为 js 即可。

好的了解