immortalwrt/immortalwrt #1056: UBNT ER-X 编译错误

lwb1978, 2023-09-06:
### Describe the bug

云编译UBNT ER-X时报错，错误代码片段为：
/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.130/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address    -o /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_jcg_q20.dtb /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_jcg_q20.dtb.tmp
Error: ../dts/mt7621_jcg_q20.dtsi:116.1-9 Label or path factory not found
FATAL ERROR: Syntax error parsing input tree
make[5]: Leaving directory '/workdir/openwrt/target/linux/ramips/image'
make[5]: *** [Makefile:235: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_jcg_q20.dtb] Error 1
make[4]: *** [Makefile:22: install] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/ramips'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/install#95.75#13.05#105.90
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:24: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:232: world] Error 2
Error: Process completed with exit code 2.

### ImmortalWrt version

5.15

### ImmortalWrt target/subtarget

main

### Device

ubnt-erx

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

xlighting2017, 2023-09-06:
老哥你配置文件不对

具体哪里不对， 你得先给出配置文件

但是原因绝对是配置文件不对

1715173329, 2023-09-06:
应该是这个 commit 的问题： 96836faf8a2e34d2e7ff9f1cacd8d2e422d906f9
明天再修吧，先摸了

lwb1978, 2023-09-07:
> 老哥你配置文件不对
> 
> 具体哪里不对， 你得先给出配置文件
> 
> 但是原因绝对是配置文件不对

以下是我的配置文件，能否帮看看哪里有错吗？不过老大说可能是mt7621_jcg_q20现在commmit问题，看今天如果老大能修的话我再重新编译看看。
[config.txt](https://github.com/immortalwrt/immortalwrt/files/12543391/config.txt)

commit c56104b9 (2023-09-07): ramips: fix reference of factory node for jcg q20

Fixes: #1056
commit 5fe28883 (2023-09-07): ramips: fix reference of factory node for jcg q20

Fixes: #1056

(cherry picked from commit c56104b985f01909c81d8dc5c126577615da91c8)
closed 2023-09-07 by 1715173329