immortalwrt/immortalwrt #1060: ImmortalWrt 23.05.0-rc3上open clash的问题

ForestL18, 2023-09-09:
### Describe the bug

设备R4S ，固件ImmortalWrt 23.05.0-rc3，在使用openclash插件时会出现域名解析延迟的现象。

具体表现为假如开一把王者，在匹配队友成功等待所有人确认时，会出现画面显示10个人全部确认但没有跳转到选择英雄界面，这时候游戏会提示尝试重连，等待十几秒左右才能重连成功，进入英雄选择界面后会发现已经过去了20秒左右。

通过查看open clash的连接信息来看，应该是[tiem-cdn.qq.com:443](url)没有及时解析成功导致。在游戏尝试重连的时候解析成功就顺利跳转了界面。

之所以到immortalwrt这来提issue，是因为通过试验发现，在相同的open clash配置的情况下，使用istoreos时，上述情况未发生，只有在使用immortalwrt时发生，故排除open clash问题。

### ImmortalWrt version

ImmortalWrt 23.05.0-rc3

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

R4S

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

ForestL18, 2023-09-09:
上述的immortalwrt和istoreos仅修改网口的IP，设置了openclash，其余设置为默认

1715173329, 2023-09-09:
推荐继续用 istoreos，这里不是论坛。

closed 2023-09-09 by 1715173329