immortalwrt/immortalwrt #1065: R68S 固件编译错误

lwb1978, 2023-09-12:
### Describe the bug

错误代码片段：

Collected errors:
Configuring shadowsocks-rust-sslocal.
 * check_conflicts_for: The following packages conflict with ntfs3-mount:
Configuring shadowsocks-libev-ss-local.
 * check_conflicts_for: 	ntfs-3g * 
Configuring omcproxy.
 * opkg_install_cmd: Cannot install package automount.
Configuring luci-app-omcproxy.
 * check_conflicts_for: The following packages conflict with ntfs3-mount:
Configuring luci-i18n-omcproxy-zh-cn.
 * check_conflicts_for: 	ntfs-3g * 
Configuring kmod-usb-net-cdc-ether.
 * opkg_install_cmd: Cannot install package ntfs3-mount.
Configuring kmod-usb-net-cdc-ncm.
make[2]: *** [package/Makefile:70: package/install] Error 255
Configuring kmod-fs-ntfs3.
Configuring simple-obfs-client.
Configuring ebtables-nft.
Configuring shadowsocksr-libev-ssr-local.
Configuring luci-i18n-ddns-zh-cn.
Configuring smartdns.
Configuring luci-app-smartdns.
Configuring luci-i18n-smartdns-zh-cn.
Configuring kmod-fuse.
Configuring ntfs-3g.
Configuring libcares.
Configuring libopenssl-conf.
Configuring luci-app-wol.
Configuring luci-app-samba4.
Configuring kmod-usb-net-rtl8150.
Configuring r8152-firmware.
Configuring kmod-usb-net-rtl8152.
Configuring luci-i18n-wol-zh-cn.
Configuring luci-app-upnp.
Configuring kmod-loop.
Configuring shadowsocks-libev-ss-redir.
Configuring default-settings.
Configuring luci-i18n-base-zh-cn.
Configuring default-settings-chn.
Configuring luci-i18n-samba4-zh-cn.
Configuring luci-i18n-ttyd-zh-cn.
Configuring coreutils-nohup.
Configuring tcping.
Configuring brook.
Configuring shadowsocksr-libev-ssr-redir.
Configuring xray-core.
Configuring luci-app-passwall.
Configuring luci-i18n-passwall-zh-cn.
Configuring fdisk.
Configuring kmod-scsi-generic.
Configuring kmod-usb-uhci.
Configuring shadowsocks-libev-ss-server.
Configuring luci-i18n-opkg-zh-cn.
Configuring kmod-lib-lzo.
Configuring luci-i18n-upnp-zh-cn.
make[1]: *** [package/Makefile:115: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_install] Error 2
Configuring kmod-usb-storage-extras.
Configuring mount-utils.
Configuring vsftpd.
Configuring urngd.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring dnsmasq-full.
make[2]: Leaving directory '/workdir/openwrt'
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:232: world] Error 2
Error: Process completed with exit code 2.

### ImmortalWrt version

6.1

### ImmortalWrt target/subtarget

master/target/linux /rockchip

### Device

R68S

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-09-12:
ntfs-3g 与 ntfs3-mount 冲突。

closed 2023-09-12 by 1715173329

lwb1978, 2023-09-12:
感谢，我删除掉ntfs-3g试试

1715173329, 2023-09-12:
https://blog.imouto.in/#/posts/8
我们已提供 ImageBuilder，不需要滥用 GitHub Actions 做完整构建。