immortalwrt/immortalwrt #1077: 21.02 的 wget-ssl 编译失败

Gzxhwq, 2023-10-01:
### Describe the bug

```
make[7]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/wget-ssl/wget-1.21.3/lib'
ccache_cc -DHAVE_CONFIG_H -I. -I../src  -DDEFAULT_TEXT_DOMAIN=\"wget-gnulib\" -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-10.2.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-10.2.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-10.2.0_musl/include  -I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/include   -Wno-cast-qual -Wno-conversion -Wno-float-equal -Wno-sign-compare -Wno-undef -Wno-unused-function -Wno-unused-parameter -Wno-float-conversion -Wimplicit-fallthrough -Wno-pedantic -Wno-sign-conversion -Wno-type-limits -Wno-unsuffixed-float-constants -I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/include  -I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/include  -DNDEBUG -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/wget-ssl/wget-1.21.3=wget-1.21.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -MT libgnu_a-openat-proc.o -MD -MP -MF .deps/libgnu_a-openat-proc.Tpo -c -o libgnu_a-openat-proc.o `test -f 'openat-proc.c' || echo './'`openat-proc.c
In file included from /workdir/openwrt/staging_dir/toolchain-x86_64_gcc-10.2.0_musl/include/endian.h:23,
                 from /workdir/openwrt/staging_dir/toolchain-x86_64_gcc-10.2.0_musl/include/sys/types.h:70,
                 from ./sys/types.h:39,
                 from ./unistd.h:141,
                 from ./stdlib.h:100,
                 from openat-priv.h:25,
                 from openat-proc.c:22:
./stdint.h:89:5: error: #if with no expression
   89 | #if
      |     ^
make[7]: *** [Makefile:3022: libgnu_a-openat-proc.o] Error 1
make[7]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/wget-ssl/wget-1.21.3/lib'
make[6]: *** [Makefile:2528: all] Error 2
make[6]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/wget-ssl/wget-1.21.3/lib'
make[5]: *** [Makefile:1741: all-recursive] Error 1
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/wget-ssl/wget-1.21.3'
make[4]: *** [Makefile:1693: all] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/wget-ssl/wget-1.21.3'
make[3]: *** [Makefile:115: /workdir/openwrt/build_dir/target-x86_64_musl/wget-ssl/wget-1.21.3/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/wget'
time: package/feeds/packages/wget/ssl/compile#0.33#0.10#0.41
    ERROR: package/feeds/packages/wget failed to build (build variant: ssl).
make[2]: *** [package/Makefile:120: package/feeds/packages/wget/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:114: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
```

### ImmortalWrt version

21.02 最新

### ImmortalWrt target/subtarget

x86/64

### Device

x86_64

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

Vyacheslav-S, 2023-10-01:
I confirm, a similar error occurred when building the package.
How can this be fixed?

Vyacheslav-S, 2023-10-01:
Now I have rolled back the package version to 1.21.1:
https://github.com/immortalwrt/packages/tree/7bafa31120b78bd10a3bfb53097cba2e0c7e2faa/net/wget
The wget-ssl package was assembled without errors.

1715173329, 2023-10-03:
Now I can reproduce this error on one of our buildbot. Let me check if there's a simple way to fix it or just revert.

1715173329, 2023-10-04:
should be fixed now

closed 2023-10-04 by 1715173329