immortalwrt/immortalwrt #1080: rock 5b: doesn't boot on latest master

alexeys85, 2023-10-04:
### Describe the bug

Tried to boot from `immortalwrt-rockchip-armv8-radxa_rock-5b-ext4-sysupgrade.img.gz` but failed.
<details><summary>Boot log</summary>
channel[0] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=16 Size=4096MB
channel[1] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=16 Size=4096MB
channel[2] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=16 Size=4096MB
channel[3] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=16 Size=4096MB
Manufacturer ID:0xff
CH0 RX Vref:27.5%, TX Vref:21.8%,22.8%
CH1 RX Vref:28.8%, TX Vref:22.8%,21.8%
CH2 RX Vref:27.9%, TX Vref:24.8%,23.8%
CH3 RX Vref:29.7%, TX Vref:21.8%,21.8%
change to F1: 528MHz
change to F2: 1068MHz
change to F3: 1560MHz
change to F0: 2112MHz
out

U-Boot SPL 2023.10-rc3-ImmortalWrt-r27718-1f9fc9a317 (Sep 26 2023 - 07:56:44 +0000)
Trying to boot from MMC2
INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-627-gb7c5102a2:derrick.huang
NOTICE:  BL31: Built : 14:52:40, Aug 10 2023
INFO:    spec: 0x1
INFO:    ext 32k is not valid
INFO:    ddr: stride-en 4CH
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    valid_cpu_msk=0xff bcore0_rst = 0x0, bcore1_rst = 0x0
INFO:    l3 cache partition cfg-0
INFO:    system boots from cpu-hwid-0
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
INFO:    dfs DDR fsp_params[0].freq_mhz= 2112MHz
INFO:    dfs DDR fsp_params[1].freq_mhz= 528MHz
INFO:    dfs DDR fsp_params[2].freq_mhz= 1068MHz
INFO:    dfs DDR fsp_params[3].freq_mhz= 1560MHz
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0xa00000
INFO:    SPSR = 0x3c9

U-Boot 2023.10-rc3-ImmortalWrt-r27718-1f9fc9a317 (Sep 26 2023 - 07:56:44 +0000)

Model: Radxa ROCK 5 Model B
DRAM:  16 GiB (effective 15.7 GiB)
Core:  329 devices, 29 uclasses, devicetree: separate
MMC:   mmc@fe2c0000: 1, mmc@fe2e0000: 0
Loading Environment from nowhere... OK
In:    serial@feb50000
Out:   serial@feb50000
Err:   serial@feb50000
Model: Radxa ROCK 5 Model B
Net:
Warning: eth_rtl8169 MAC addresses don't match:
Address in ROM is               00:e0:4c:68:04:3d
Address in environment is       7a:65:b3:2a:03:82
eth0: eth_rtl8169
Hit any key to stop autoboot:  0
** Booting bootflow 'mmc@fe2c0000.bootdev.part_1' with script
6077944 bytes read in 512 ms (11.3 MiB/s)
## Loading kernel from FIT Image at 02080000 ...
   Using 'config-1' configuration
   Verifying Hash Integrity ... OK
   Trying 'kernel-1' kernel subimage
     Description:  ARM64 OpenWrt Linux-6.1.55
     Type:         Kernel Image
[... 533 lines cut ...]
[    3.018123] hub 10-0:1.0: 1 port detected
[    3.021300] Waiting for root device PARTUUID=5452574f-02...
[   13.026381] i2c 1-0022: deferred probe pending
[   33.757829] vdd_gpu_s0: disabling
</details> 

### ImmortalWrt version

master

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

rock 5b

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-10-04:
> [ 1.622831] mmc_host mmc1: Bus speed (slot 0) = 49500000Hz (slot req 50000000Hz, actual 49500000HZ div = 0)
[ 1.626827] I/O error, dev mmcblk1, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
[ 1.628152] I/O error, dev mmcblk1, sector 1 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
[ 1.629422] I/O error, dev mmcblk1, sector 2 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
[ 1.630700] I/O error, dev mmcblk1, sector 3 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
[ 1.632022] I/O error, dev mmcblk1, sector 4 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
[ 1.633285] I/O error, dev mmcblk1, sector 5 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
[ 1.634617] I/O error, dev mmcblk1, sector 6 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
[ 1.635942] I/O error, dev mmcblk1, sector 7 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
[ 1.636698] Buffer I/O error on dev mmcblk1, logical block 0, async page read
[ 1.642601] I/O error, dev mmcblk1, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
[ 1.643862] I/O error, dev mmcblk1, sector 1 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
[ 1.648018] Buffer I/O error on dev mmcblk1, logical block 0, async page read
[ 1.648733] mmcblk1: unable to read partition table

Check your TF card.

alexeys85, 2023-10-04:
Well, I used that card to boot armbian and it worked fine. Anyway, will try another one.

alexeys85, 2023-10-05:
Tried another mmc card, which boots armbian fine, no luck
<details><summary>Details</summary>
<p>
DDR V1.13 25cee80c4f cym 23/08/11-09:31:58
LPDDR4X, 2112MHz
channel[0] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=16 Size=4096MB
channel[1] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=16 Size=4096MB
channel[2] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=16 Size=4096MB
channel[3] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=16 Size=4096MB
Manufacturer ID:0xff
CH0 RX Vref:27.9%, TX Vref:21.8%,21.8%
CH1 RX Vref:28.4%, TX Vref:22.8%,21.8%
CH2 RX Vref:27.9%, TX Vref:24.8%,23.8%
CH3 RX Vref:29.7%, TX Vref:21.8%,21.8%
change to F1: 528MHz
change to F2: 1068MHz
change to F3: 1560MHz
change to F0: 2112MHz
out

U-Boot SPL 2023.10-rc3-ImmortalWrt-r27718-1f9fc9a317 (Sep 26 2023 - 07:56:44 +0000)
Trying to boot from MMC2
INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-627-gb7c5102a2:derrick.huang
NOTICE:  BL31: Built : 14:52:40, Aug 10 2023
INFO:    spec: 0x1
INFO:    ext 32k is not valid
INFO:    ddr: stride-en 4CH
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    valid_cpu_msk=0xff bcore0_rst = 0x0, bcore1_rst = 0x0
INFO:    l3 cache partition cfg-0
INFO:    system boots from cpu-hwid-0
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
INFO:    dfs DDR fsp_params[0].freq_mhz= 2112MHz
INFO:    dfs DDR fsp_params[1].freq_mhz= 528MHz
INFO:    dfs DDR fsp_params[2].freq_mhz= 1068MHz
INFO:    dfs DDR fsp_params[3].freq_mhz= 1560MHz
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0xa00000
INFO:    SPSR = 0x3c9

U-Boot 2023.10-rc3-ImmortalWrt-r27718-1f9fc9a317 (Sep 26 2023 - 07:56:44 +0000)

Model: Radxa ROCK 5 Model B
DRAM:  16 GiB (effective 15.7 GiB)
Core:  329 devices, 29 uclasses, devicetree: separate
MMC:   mmc@fe2c0000: 1, mmc@fe2e0000: 0
Loading Environment from nowhere... OK
In:    serial@feb50000
Out:   serial@feb50000
Err:   serial@feb50000
Model: Radxa ROCK 5 Model B
Net:
Warning: eth_rtl8169 MAC addresses don't match:
Address in ROM is               00:e0:4c:68:04:3d
Address in environment is       7a:65:b3:2a:03:82
eth0: eth_rtl8169
Hit any key to stop autoboot:  0
** Booting bootflow 'mmc@fe2c0000.bootdev.part_1' with script
6077944 bytes read in 506 ms (11.5 MiB/s)
## Loading kernel from FIT Image at 02080000 ...
   Using 'config-1' configuration
   Verifying Hash Integrity ... OK
   Trying 'kernel-1' kernel subimage
     Description:  ARM64 OpenWrt Linux-6.1.55
[... 515 lines cut ...]
[    2.935623] pci 0002:20:00.0: supports D1 D2
[    2.935999] pci 0002:20:00.0: PME# supported from D0 D1 D3hot
[    2.938669] pci 0002:20:00.0: Primary bus is hard wired to 0
[    2.939167] pci 0002:20:00.0: bridge configuration invalid ([bus 01-ff]), reconfiguring
[    2.939980] pci_bus 0002:21: busn_res: [bus 21-2f] end is updated to 21
[    2.940569] pci 0002:20:00.0: BAR 6: assigned [mem 0xf2200000-0xf220ffff pref]
[    2.941205] pci 0002:20:00.0: PCI bridge to [bus 21]
[    2.942801] pcieport 0002:20:00.0: PME: Signaling with IRQ 112
[    2.943807] pcieport 0002:20:00.0: AER: enabled with IRQ 113
[    2.954420] ehci-platform fc880000.usb: EHCI Host Controller
[    2.954474] ohci-platform fc8c0000.usb: Generic Platform OHCI controller
[    2.954925] ehci-platform fc880000.usb: new USB bus registered, assigned bus number 9
[    2.955508] ohci-platform fc8c0000.usb: new USB bus registered, assigned bus number 10
[    2.956234] ehci-platform fc880000.usb: irq 78, io mem 0xfc880000
[    2.956913] ohci-platform fc8c0000.usb: irq 80, io mem 0xfc8c0000
[    2.973545] ehci-platform fc880000.usb: USB 2.0 started, EHCI 1.00
[    2.974279] hub 9-0:1.0: USB hub found
[    2.974618] hub 9-0:1.0: 1 port detected
[    3.021745] hub 10-0:1.0: USB hub found
[    3.022096] hub 10-0:1.0: 1 port detected
[    3.024613] Waiting for root device PARTUUID=5452574f-02...
[   13.026687] i2c 1-0022: deferred probe pending
[   33.757783] vdd_gpu_s0: disabling
</p>
</details>

alexeys85, 2023-10-05:
Successful boot original openwrt with this card on nanopi R4S

alexeys85, 2023-10-05:
Part of boot log from armbian:
```
[    1.564173] dwmmc_rockchip fe2c0000.mmc: IDMAC supports 32-bit address mode.
[    1.564196] dwmmc_rockchip fe2c0000.mmc: Using internal DMA controller.
[    1.564202] dwmmc_rockchip fe2c0000.mmc: Version ID is 270a
[    1.564215] dwmmc_rockchip fe2c0000.mmc: DW MMC controller at irq 71,32 bit host data width,256 deep fifo
[    1.564510] dwmmc_rockchip fe2c0000.mmc: Got CD GPIO
[    1.577194] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[    1.614502] mmc_host mmc1: Bus speed (slot 0) = 49500000Hz (slot req 50000000Hz, actual 49500000HZ div = 0)
[    1.614691] mmc1: new high speed SDHC card at address 211f
[    1.615315] mmcblk1: mmc1:211f APPSD 29.5 GiB
[    1.621993] GPT:Primary header thinks Alt. header is not at the end of the disk.
[    1.622002] GPT:2850815 != 61849599
[    1.622006] GPT:Alternate GPT header not at the end of the disk.
[    1.622008] GPT:2850815 != 61849599
[    1.622010] GPT: Use GNU Parted to correct GPT errors.
[    1.622021]  mmcblk1: p1 p2
[    3.614008] EXT4-fs (mmcblk1p2): mounted filesystem b4d312c2-c3a9-49a3-8cf7-02dc202c2a90 ro with writeback data mode. Quota mode: none.
```

1715173329, 2023-10-07:
Fixed now.

commit f1346bf7 (2023-10-07): rockchip: refresh upstreamed patches

Fixes: #1080
closed 2023-10-07 by 1715173329