immortalwrt/immortalwrt #1082: zyxel ex5601-t0 crashes

SuperKali, 2023-10-11:
### Describe the bug

Hello,

I don't know the real problem, but i started to configure vlan for internet and it crashed every time.

### ImmortalWrt version

Latest

### ImmortalWrt target/subtarget

Latest

### Device

zyxel ex5601-t0

### Image kind

Official downloaded image

### Steps to reproduce

Adding VLAN for internet 

### Additional info

root@ImmortalWrt:/# [  121.598622] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
[  121.605387] br-lan: port 5(phy1-ap0) entered blocking state
[  121.610973] br-lan: port 5(phy1-ap0) entered forwarding state
[  170.587282] Unable to handle kernel access to user memory outside uaccess routines at virtual address 0000000000000008
[  170.597982] Mem abort info:
[  170.600765]   ESR = 0x0000000096000005
[  170.604497]   EC = 0x25: DABT (current EL), IL = 32 bits
[  170.609813]   SET = 0, FnV = 0
[  170.612856]   EA = 0, S1PTW = 0
[  170.615981]   FSC = 0x05: level 1 translation fault
[  170.620849] Data abort info:
[  170.623715]   ISV = 0, ISS = 0x00000005
[  170.627538]   CM = 0, WnR = 0
[  170.630492] user pgtable: 4k pages, 39-bit VAs, pgdp=000000004757f000
[  170.636921] [0000000000000008] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[  170.645602] Internal error: Oops: 96000005 [#1] SMP
[  170.650464] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_mppe ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_fullcone nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 cfg80211 ts_kmp ts_fsm ts_bm slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcrc32c crc_ccitt compat cls_flower asn1_decoder act_vlan crypto_safexcel cls_bpf act_bpf sch_tbf sch_ingress sch_htb
[  170.650593]  sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact cryptodev sit tunnel4 ip_tunnel crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_generic seqiv md5 des_generic libdes authencesn authenc arc4 leds_gpio gpio_button_hotplug
[  170.765218] CPU: 1 PID: 1684 Comm: hostapd Not tainted 5.15.132 #0
[  170.771381] Hardware name: Zyxel EX5601-T0 (DT)
[  170.775894] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  170.782834] pc : sta_set_sinfo+0xa94/0xd60 [mac80211]
[  170.787903] lr : sta_set_sinfo+0x5fc/0xd60 [mac80211]
[  170.792953] sp : ffffffc00aa73780
[  170.796251] x29: ffffffc00aa73780 x28: 0000000000000001 x27: ffffffc00aa73dd0
[  170.803367] x26: ffffff8000011080 x25: ffffff8007ef4900 x24: ffffff8006b92735
[  170.810483] x23: ffffffc00aa73a38 x22: ffffff8006b92aa8 x21: 0000058d1783bf82
[  170.817598] x20: ffffff8006b92000 x19: ffffff80021a7400 x18: 0000000000000000
[  170.824713] x17: 0000000000000fc0 x16: ffffffc0095fe000 x15: 00000000000007e0
[  170.831828] x14: 000000000000038d x13: 0000000000000001 x12: 0000000000000000
[  170.838943] x11: 0000000000000001 x10: 0000000000000900 x9 : ffffffc00aa734f0
[  170.846059] x8 : ffffff80073cd760 x7 : ffffff8003f70880 x6 : 00000000000000c0
[  170.853174] x5 : ffffff8003f703a0 x4 : 0000000000000000 x3 : 0000000000000000
[  170.860288] x2 : ffffff80021a7444 x1 : 0000000000000001 x0 : 0000000000000000
[  170.867404] Call trace:
[  170.869836]  sta_set_sinfo+0xa94/0xd60 [mac80211]
[  170.874548]  sta_set_sinfo+0xbdc/0xd60 [mac80211]
[  170.879252]  sta_info_destroy_addr_bss+0x4c/0x70 [mac80211]
[  170.884824]  ieee80211_color_change_finish+0x1578/0x17e0 [mac80211]
[  170.891089]  cfg80211_bss_color_notify+0x394/0x484 [cfg80211]
[  170.896838]  genl_family_rcv_msg_doit+0xb4/0x110
[  170.901441]  genl_rcv_msg+0xd0/0x1c0
[  170.905001]  netlink_rcv_skb+0x58/0x120
[  170.908824]  genl_rcv+0x34/0x50
[  170.911949]  netlink_unicast+0x1f0/0x2ec
[  170.915856]  netlink_sendmsg+0x19c/0x3d0
[  170.919764]  ____sys_sendmsg+0x27c/0x2b0
[  170.923674]  ___sys_sendmsg+0x80/0xf0
[  170.927322]  __sys_sendmsg+0x44/0xa0
[  170.930881]  __arm64_sys_sendmsg+0x20/0x30
[  170.934959]  invoke_syscall.constprop.0+0x4c/0xe0
[  170.939649]  do_el0_svc+0x40/0xd0
[  170.942949]  el0_svc+0x14/0x50
[  170.945993]  el0t_64_sync_handler+0xe0/0x110
[  170.950245]  el0t_64_sync+0x15c/0x160
[  170.953896] Code: d3441c63 12000c00 8b030ca3 f9409c63 (f9400465) 
[  170.959971] ---[ end trace ea1a8a947017c3f3 ]---
[  170.968365] Kernel panic - not syncing: Oops: Fatal exception
[  170.974095] SMP: stopping secondary CPUs
[  170.978004] Kernel Offset: disabled
[  170.981474] CPU features: 0x0,00000000,20000802
[  170.985988] Memory Limit: none
[  170.992702] Rebooting in 3 seconds..

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

SuperKali, 2023-10-11:
Sorry, i think is a radio problem because on stack trace i see a lot of ieee80211

1715173329, 2023-10-15:
I don't have this device, it's added by upstream. 
This issue should be route to there as well.

SuperKali, 2023-12-20:
hello @1715173329, i have tried the openwrt stock, this problem is not present

1715173329, 2024-01-06:
https://github.com/immortalwrt/immortalwrt/blob/342a17476c73575e9e94a0178eadb4fc1e78ad7a/package/kernel/mt76/Makefile#L241

Disable WED support and try again.
You can also try to remove `bridger` package.

1715173329, 2024-03-04:
Closed due to no response.

closed 2024-03-04 by 1715173329