immortalwrt/immortalwrt #1094: 希望支持360 T7新版flash W25N01KV

yidianyidian, 2023-10-18:
### Describe the bug

如题，希望支持360 T7新版flash W25N01KV，目前的固件刷入后无法开机。可能相关的信息https://github.com/hanwckf/immortalwrt-mt798x/commit/a65e09d263825aea948f9b19954841cca1dfb946

### ImmortalWrt version

all

### ImmortalWrt target/subtarget

all

### Device

360 T7

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

tarly, 2023-11-03:
Is this chip supported in snapshot now ?

1715173329, 2023-11-17:
已支持。

closed 2023-11-17 by 1715173329