immortalwrt/immortalwrt #1096: ZTE e8820s 无5G

drx1x2, 2023-10-22:
### Describe the bug

Sun Oct  1 10:22:05 2023 kern.notice kernel: [    0.000000] Linux version 5.15.132 (buildbot@buildbot.infra.immortalwrt.org) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r27086-d9bdd43aa7) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Sun Oct 1 02:21:58 2023
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] MIPS: machine is ZTE E8820S
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sun Oct  1 10:22:05 2023 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] Zone ranges:
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] Movable zone start for each node
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] Early memory node ranges
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] percpu: Embedded 11 pages/cpu s14736 r8192 d22128 u45056
Sun Oct  1 10:22:05 2023 kern.debug kernel: [    0.000000] pcpu-alloc: s14736 r8192 d22128 u45056 alloc=11*4096
Sun Oct  1 10:22:05 2023 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
Sun Oct  1 10:22:05 2023 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] Writing ErrCtl register=0000040f
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] Readback ErrCtl register=0000040f
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] Memory: 248480K/262144K available (7308K kernel code, 624K rwdata, 1488K rodata, 1256K init, 226K bss, 13664K reserved, 0K cma-reserved)
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] NR_IRQS: 256
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.008050] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.066221] pid_max: default: 32768 minimum: 301
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.071617] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.078828] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.090361] rcu: Hierarchical SRCU implementation.
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.095818] smp: Bringing up secondary CPUs ...
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.101202] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.101229] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sun Oct  1 10:22:05 2023 kern.warn kernel: [    0.101244] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.101288] CPU1 revision is: 0001992f (MIPS 1004Kc)
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.160591] Synchronize counters for CPU 1: done.
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.193109] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.193130] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sun Oct  1 10:22:05 2023 kern.warn kernel: [    0.193141] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.193168] CPU2 revision is: 0001992f (MIPS 1004Kc)
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.252074] Synchronize counters for CPU 2: done.
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.282690] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.282711] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sun Oct  1 10:22:05 2023 kern.warn kernel: [    0.282722] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.282753] CPU3 revision is: 0001992f (MIPS 1004Kc)
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.337256] Synchronize counters for CPU 3: done.
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.367113] smp: Brought up 1 node, 4 CPUs
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.376358] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.386162] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.393235] pinctrl core: initialized pinctrl subsystem
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.400081] NET: Registered PF_NETLINK/PF_ROUTE protocol family
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.406695] thermal_sys: Registered thermal governor 'step_wise'
Sun Oct  1 10:22:05 2023 kern.debug kernel: [    0.418115] FPU Affinity set after 11720 emulations
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.436759] clocksource: Switched to clocksource GIC
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.443171] NET: Registered PF_INET protocol family
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.448221] IP idents hash table entries: 4096 (order: 3, 32768 bytes, linear)
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.456457] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.464759] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
Sun Oct  1 10:22:05 2023 kern.info kernel: [    0.472435] TCP established hash table entries: 2048 (order: 1, 8192 bytes, linear)
[... 308 lines cut ...]
Sun Oct  1 10:22:22 2023 daemon.notice netifd: Network device 'lan2' link is up
Sun Oct  1 10:22:38 2023 daemon.warn odhcpd[2043]: No default route present, overriding ra_lifetime!
Sun Oct  1 10:22:44 2023 daemon.err uhttpd[978]: [info] luci: accepted login on / for root from 192.168.1.2
Sun Oct  1 10:22:54 2023 daemon.warn odhcpd[2043]: No default route present, overriding ra_lifetime!
Sun Oct  1 10:23:10 2023 daemon.warn odhcpd[2043]: No default route present, overriding ra_lifetime!

### ImmortalWrt version

21 / 23

### ImmortalWrt target/subtarget

null

### Device

zte E8820S

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-10-22:
I don't know why it didn't work, there's no error in your log.

drx1x2, 2023-10-24:
https://openwrt.ai/?target=ramips%2Fmt7621&id=zte_e8820s

生成的固件，5G正常

1715173329, 2023-10-24:
那你就去用这个即可。

closed 2023-10-24 by 1715173329