immortalwrt/immortalwrt #1103: Firmware Selector custom build Snapshot Generic x86/64 failed

Ricky-Hao, 2023-11-06:
### Describe the bug

Trigger build with default packages on Firmware Selector: [https://firmware-selector.immortalwrt.org/?version=SNAPSHOT&target=x86%2F64&id=generic
](https://firmware-selector.immortalwrt.org/?version=SNAPSHOT&target=x86%2F64&id=generic
)

And build return error.
[image]

### ImmortalWrt version

r28008-3f9bf2c41b

### ImmortalWrt target/subtarget

x86/64

### Device

Microsoft Corporation Virtual Machine

### Image kind

Official downloaded image

### Steps to reproduce

Trigger build with default packages on Firmware Selector: [https://firmware-selector.immortalwrt.org/?version=SNAPSHOT&target=x86%2F64&id=generic
](https://firmware-selector.immortalwrt.org/?version=SNAPSHOT&target=x86%2F64&id=generic
)

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-11-06:
upstream issue, still under investigation

1715173329, 2023-11-07:
Upstream fixed.

closed 2023-11-07 by 1715173329