immortalwrt/immortalwrt #1108: 编译的x86，启动的时候报一堆这个错

asiaon123, 2023-11-10:
### Describe the bug

## 系统日志

```
[    4.676205] procd: - ubus -
[    4.838434] procd: - init -
[    4.928379] kmodloader: loading kernel modules from /etc/modules.d/*
[    4.935833] BPF: 	 type_id=55 bits_offset=512
[    4.936486] BPF:  
[    4.936895] BPF: Invalid name
[    4.937379] BPF: 
[    4.937779] failed to validate module [crypto_acompress] BTF: -22
[    4.941046] BPF: 	4dval type_id=110044 bits_offset=0
[    4.941712] BPF:  
[    4.942108] BPF: Invalid name
[    4.942574] BPF: 
[    4.942953] failed to validate module [af_alg] BTF: -22
[    4.943921] NET: Registered PF_ALG protocol family
[    4.945118] BPF: 	4dval type_id=110039 bits_offset=0
[    4.945771] BPF:  
[    4.946149] BPF: Invalid name
[    4.946624] BPF: 
[    4.946988] failed to validate module [algif_aead] BTF: -22
[    4.948313] BPF: 	4dval type_id=110039 bits_offset=0
[    4.948956] BPF:  
[    4.949327] BPF: Invalid name
[    4.949799] BPF: 
[    4.950153] failed to validate module [algif_hash] BTF: -22
[    4.951754] BPF: 	4dval type_id=110039 bits_offset=0
[    4.952359] BPF:  
[    4.952719] BPF: Invalid name
[    4.953159] BPF: 
[    4.953510] failed to validate module [algif_skcipher] BTF: -22
[    4.956402] BPF: 	 type_id=110053 bits_offset=0
[    4.956956] BPF:  
[    4.957298] BPF: Invalid name
[    4.957726] BPF: 
[    4.958062] failed to validate module [ena] BTF: -22
[    4.959518] BPF: 	 type_id=2333 bits_offset=0
[    4.960061] BPF:  
[    4.960405] BPF: Invalid name
[    4.960817] BPF: 
[    4.961145] failed to validate module [ip6_udp_tunnel] BTF: -22
[    4.962441] BPF: 	 type_id=2333 bits_offset=0
[    4.962974] BPF:  
[    4.963307] BPF: Invalid name
[    4.963739] BPF: 
[    4.964072] failed to validate module [udp_tunnel] BTF: -22
[    4.965370] BPF: 	8253_init type_id=55128 bitfield_size=0 bits_offset=640
[    4.966135] BPF:  
[    4.966474] BPF: Invalid name
[    4.966886] BPF: 
[    4.967215] failed to validate module [vxlan] BTF: -22
[    4.969767] BPF: 	 type_id=70 bits_offset=16840
[    4.970322] BPF:  
[    4.970662] BPF: Invalid name
[    4.971071] BPF: 
[    4.971394] failed to validate module [snd] BTF: -22
[    4.972610] BPF: 	 type_id=70 bits_offset=16840
[    4.973169] BPF:  
[    4.973524] BPF: Invalid name
[    4.973957] BPF: 
[    4.974279] failed to validate module [snd_timer] BTF: -22
[    4.975546] urngd: v1.0.2 started.
[    4.976010] BPF: 	 type_id=70 bits_offset=16840
[    4.976555] BPF:  
[    4.976889] BPF: Invalid name
[    4.977294] BPF: 
[    4.977630] failed to validate module [snd_pcm] BTF: -22
[... 537 lines cut ...]
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-variant-release=y
```

### ImmortalWrt version

r28047-fa6d637512

### ImmortalWrt target/subtarget

x86/64

### Device

QEMU Standard PC (i440FX + PIIX, 1996)

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-11-10:
如果开启 btf，每次必须全新编译。
此错误不会影响正常使用。

closed 2023-11-10 by 1715173329

asiaon123, 2023-11-10:
我没有主动开启这个，我看到的副作用是

[image]

流量统计的不能正常工作