immortalwrt/immortalwrt #1111: 23.05分支X86-64固件，升级时无法保留配置

nantayo, 2023-11-12:
### Describe the bug

最近我将一台小主机的 Immortalwrt 大版本从 21.02-snapshot 迁移到 23.05-snapshot，在一般情况下它运行的非常好。
但我注意到对于 23.05 版本，当我在Web页面刷写固件时，无论是否选择保留配置，刷写完毕后都会丢失所有的配置（甚至包括该系统磁盘上额外的分区表信息）。
注：刷写指的是从较旧的 23.05 版本升级到较新的 23.05 版本。

我想知道这是否是某个已知的问题或者特性，以及是否有方法规避它？

### ImmortalWrt version

r0-32885f2

### ImmortalWrt target/subtarget

x86/64

### Device

CncTion N4505-4L

### Image kind

Self-built image

### Steps to reproduce

使用Web页面刷写新的固件（无论勾选其中的任何选项都一样）。

### Actual behaviour

系统本身可以运行，可以正常保存配置（重启也不会丢失），但升级刷写后会丢失配置。

### Expected behaviour

希望升级时能够正常保留配置。

### Additional info

1. 我使用 Github Actions 编译自己的固件，每次编译都应当是全新的，不会受到旧文件的影响。
2. 虽然磁盘的分区表会丢失，但存储的文件却不会。我使用 PE 系统下的 Diskgenius 软件找回丢失的分区（ext4）后，上面的文件都回来了，没有错误或者丢失。
3. 我使用 df -h 命令查看，各个分区的可用空间都是充足的。
[image]

### Diffconfig

```text
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_TARGET_ROOTFS_EXT4FS=n
CONFIG_GRUB_IMAGES=n
CONFIG_GRUB_CONSOLE=n
CONFIG_GRUB_TIMEOUT="0"
CONFIG_VMDK_IMAGES=n
CONFIG_TARGET_KERNEL_PARTSIZE=12
CONFIG_TARGET_ROOTFS_PARTSIZE=500

# ipv6
CONFIG_PACKAGE_ip6tables-extra=y
CONFIG_PACKAGE_ip6tables-mod-nat=y
CONFIG_PACKAGE_ipv6helper=y

# luci
CONFIG_PACKAGE_luci-app-dockerman=y
CONFIG_PACKAGE_luci-app-filetransfer=n
CONFIG_PACKAGE_luci-app-mosdns=y
CONFIG_PACKAGE_luci-app-passwall=y
CONFIG_PACKAGE_luci-app-qbittorrent=y
CONFIG_PACKAGE_luci-app-samba4=y
CONFIG_PACKAGE_luci-app-turboacc=n
CONFIG_PACKAGE_luci-app-zerotier=y

# kmod
CONFIG_PACKAGE_kmod-8139cp=n
CONFIG_PACKAGE_kmod-8139too=n
CONFIG_PACKAGE_kmod-amazon-ena=n
CONFIG_PACKAGE_kmod-amd-xgbe=n
CONFIG_PACKAGE_kmod-bnx2=n
CONFIG_PACKAGE_kmod-e1000=n
CONFIG_PACKAGE_kmod-e1000e=n
CONFIG_PACKAGE_kmod-forcedeth=n
CONFIG_PACKAGE_kmod-i2c-core=n
CONFIG_PACKAGE_kmod-i40e=n
CONFIG_PACKAGE_kmod-igb=n
CONFIG_PACKAGE_kmod-igbvf=n
CONFIG_PACKAGE_kmod-ixgbe=n
CONFIG_PACKAGE_kmod-ixgbevf=n
CONFIG_PACKAGE_kmod-nvme=y
CONFIG_PACKAGE_kmod-pcnet32=n
CONFIG_PACKAGE_kmod-r8101=n
CONFIG_PACKAGE_kmod-r8125=n
CONFIG_PACKAGE_kmod-r8168=n
CONFIG_PACKAGE_kmod-tg3=n
CONFIG_PACKAGE_kmod-tulip=n
CONFIG_PACKAGE_kmod-usb3=y
CONFIG_PACKAGE_kmod-usb-audio=n
CONFIG_PACKAGE_kmod-usb-net=n
CONFIG_PACKAGE_kmod-usb-net-asix=n
CONFIG_PACKAGE_kmod-usb-net-asix-ax88179=n
CONFIG_PACKAGE_kmod-usb-net-rtl8150=n
CONFIG_PACKAGE_kmod-usb-net-rtl8152-vendor=n
CONFIG_PACKAGE_kmod-vmxnet3=n

# other packages
CONFIG_PACKAGE_alsa-utils=n
CONFIG_PACKAGE_partx-utils=n
CONFIG_PACKAGE_smartmontools=y
CONFIG_PACKAGE_smartmontools-drivedb=y
CONFIG_PACKAGE_zoneinfo-asia=y
CONFIG_PACKAGE_zoneinfo-core=y
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-11-12:
Kernel 分区过小。

nantayo, 2023-11-12:
> Kernel 分区过小。

了解了，谢谢。请教kernel分区不多加kmod的话，通常应该分配多大合适

xlighting2017, 2023-11-12:
> > Kernel 分区过小。
> 
> 了解了，谢谢。请教kernel分区不多加kmod的话，通常应该分配多大合适

如果你不设置，就会有个“默认值”...

nantayo, 2023-11-12:
> Kernel 分区过小。

Hi，我想可能存在一些其他问题。
我尝试分配了 64M 的 Kernel 分区，比默认配置的 32M 还大一倍，同时我在 PE 系统中使用 Diskgenius 软件观察 Kernel 分区的用量，发现只占用了很小的 7.7M。
但依然无法在刷写时正确保留配置，还有什么可能的原因吗？

nantayo, 2023-11-13:
> 先尝试官方发行镜像，并且不要安装 docker 这类软件。 如果你确实有此类需求，请使用 pve。

好的，看起来这是一个意料之外的错误，后面有空会尝试官方镜像看看能否解决。

nantayo, 2023-11-24:
回报一下近况，用最新源码编译之后重新安装、分区，问题暂时解决了。也许是某些意外的兼容性问题导致的？
是否彻底解决有待进一步观察。

closed 2023-11-24 by nantayo