immortalwrt/immortalwrt #1113: 网页端无法更新软件包列表

shierleeee, 2023-11-13:
### Describe the bug

点击更新列表提示：无法执行 opkg update 命令：SyntaxError: Unexpected end of JSON input
在SSH可使用opkg update 更新列表，更新后刷新网页可出现软件包。

### ImmortalWrt version

r27269-32885f26cf

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Mi Router WR30U (112M UBI with NMBM-Enabled layout)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-11-13:
你网络问题，前端 timeout 了

closed 2023-11-13 by 1715173329

shierleeee, 2023-11-13:
感谢回复，是因为我在海外的原因吗？有没有解救方案？是不是换别的软件包地址就好了？

shierleeee, 2023-11-15:
> sed -i.bak "s,mirrors.vsean.net/openwrt,immortalwrt.kyarucloud.moe,g" "/etc/opkg/distfeeds.conf"

感谢，问题成功解决。