immortalwrt/immortalwrt #1116: Banana Pi R3 Mini 无法从 NAND 启动

YaeMashiro, 2023-11-16:
### Describe the bug

刷好 BL2 和 FIP 之后
TTL 显示

> F0: 102B 0000
FA: 1040 0000
FA: 1040 0000 [0200]
F9: 0000 0000
L0: 8005 0000 [0001]
00: 1012 0000
F9: 0000 0000
L0: 8005 0000 [0001]
01: 102A 0001
02: 1012 0000
BP: 2000 00C0 [0001]
EC: 0000 0000 [1000]
T0: 0000 00B8 [010F]
System halt!

emmc 正常

### ImmortalWrt version

snapshots

### ImmortalWrt release

snapshots

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

Banana Pi R3 Mini

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-11-16:
你刷的 bl2 和 fip 有误，emmc 和 nand 的不通用。

closed 2023-11-16 by 1715173329

YaeMashiro, 2023-11-16:
我刚又确认了一次，刷的是 snand

1715173329, 2023-11-16:
怎么刷的

YaeMashiro, 2023-11-16:
https://forum.banana-pi.org/t/bpi-r3-mini-review-and-mainline-immortalwrt-firmware-support/16526
你在论坛发的
不过不一样的地方是
`mtd erase BL2`
我这边是小写的 bl2 和 fip

我在 EMMC 的 ImmortalWrt 下刷机的

1715173329, 2023-11-16:
strings /dev/mtd0 | grep -E '(2023|NAND)'
strings /dev/mtd3 | grep -E '(2023|NAND)'

看下输出

YaeMashiro, 2023-11-16:
strings /dev/mtd0 | grep -E '(2023|NAND)'

> SPINAND!
Recognized SPI-NAND ID: 0x%x 0x%x 0x%x
Unrecognized SPI-NAND ID: 0x%x 0x%x 0x%x 0x%x
SPI_NAND parses attributes from parameter page.
SPI_NAND Detected ID 0x%x

strings /dev/mtd3 | grep -E '(2023|NAND)'
 > v2.9.0(release):ImmortalWrt v2023-10-13-0ea67d76-1 (mt7986-spim-nand-ddr4)
Built : 05:33:00, Nov 12 2023
[0;36m[SPI-NAND]
bootmenu_2=Boot production system from NAND.=run boot_production ; run bootmenu_confirm_return
bootmenu_3=Boot recovery system from NAND.=run boot_recovery ; run bootmenu_confirm_return
bootmenu_4=Load production system via TFTP then write to NAND.=setenv noboot 1 ; setenv replacevol 1 ; run boot_tftp_production ; setenv noboot ; setenv replacevol ; run bootmenu_confirm_return
bootmenu_5=Load recovery system via TFTP then write to NAND.=setenv noboot 1 ; setenv replacevol 1 ; run boot_tftp_recovery ; setenv noboot ; setenv replacevol ; run bootmenu_confirm_return
[31mLoad BL31+U-Boot FIP via TFTP then write to NAND.
[31mLoad BL2 preloader via TFTP then write to NAND.
Warning: Device tree includes old 'u-boot,dm-' tags: please fix by 2023.07!
Only NAND-based devices can have mark blocks
List of NAND devices:
No NAND MTD device found
Error: '%s' is not a NAND device
07/01/2023
spi-nand: %s %s: SPI NANDs with more than one die must implement ->select_target()
spi-nand: %s %s: %s SPI NAND was found.
spi-nand: %s %s: Failed to initialize the SPI NAND chip (err = %d)
NAND utility
U-Boot 2023.07.02-ImmortalWrt-r28108-b171717c5e (Nov 12 2023 - 05:33:00 +0000)
        <name>: NAND partition/chip name (or corresponding DM device name or OF path)
 > - NAND flash R/W and debugging utility
nand info - Show active NAND devices
nand select <name> - Select active NAND devices

1715173329, 2023-11-16:
BL2 没换掉啊，还是老的

YaeMashiro, 2023-11-16:
嗯？那我现在怎么做呢？

 > module is already loaded - mtd-rw
Unlocking /dev/mtd0 ...
Erasing /dev/mtd0 ...
Unlocking /dev/mtd0 ...

 > Writing from immortalwrt-mediatek-filogic-bananapi_bpi-r3-mini-snand-preloader.bin to /dev/mtd0 ...     
Unlocking /dev/mtd3 ...
Erasing /dev/mtd3 ...
Unlocking /dev/mtd3 ...

 > Writing from immortalwrt-mediatek-filogic-bananapi_bpi-r3-mini-snand-bl31-uboot.fip to /dev/mtd3 ...

1715173329, 2023-11-16:
https://transfer.sh/suodsIuiad/mt7986-bananapi_bpi-r3-mini-nand-bl2.img
https://transfer.sh/h2XYl0lHGk/mt7986-bananapi_bpi-r3-mini-nand-fip.bin

刷这两个试试呢

YaeMashiro, 2023-11-16:
> F0: 102B 0000
FA: 1040 0000
FA: 1040 0000 [0200]
F9: 0000 0000
V0: 7027 6006 [0001]
00: 1017 0000
F9: 0000 0000
V0: 7027 6006 [0001]
01: 102A 0001
02: 1017 0000
BP: 2000 00C0 [0001]
EC: 0000 0000 [1000]
T0: 0000 0396 [010F]
System halt!

新代码

1715173329, 2023-11-16:
BL2 完全烂掉了，看下你的 mtd-rw，不能用论坛里面那个。