immortalwrt/immortalwrt #1117: 23.05.1流量分载(Flow offloading)不生效导致网速变慢

microka, 2023-11-20:
### Describe the bug

[image]
J4125 i225-V(b6)*6
刷入 immortalwrt-23.05.1-x86-64-generic-squashfs-combined-efi.img.gz 固件，系统版本 ImmortalWrt 23.05.1 r27304-31bc47589e

[image]
流量分载(Flow offloading)为默认开启

只配置一条电信 2000M wan，由 ImmortalWrt PPPoE 拨号，安装 sysstat 软件包，ssh 使用 `mpstat -P ALL 1` 观察 CPU 实时占用

[image]
[电信官方IPv4测速](https://10000.gd.cn/#/speed)，下行能跑满2000M，CPU占用60-90%

[image]
[中科大IPv6测速](http://test6.ustc.edu.cn/)，下行只能跑到1300+M，CPU占用40-70%

[image]
[东北大学IPv6测速](https://speed.neu6.edu.cn/)，下行只能跑到1100+M，CPU占用40-60%

[image]
作为对照，同设备刷入 immortalwrt-21.02.7-x86-64-generic-squashfs-combined-efi.img.gz 固件，系统版本 ImmortalWrt 21.02.7 r20074-a8bbadefaf
[image]
流量分载(Flow offloading)为默认开启
其余配置与 ImmortalWrt 23.05.1 保持一致，进行测速

[image]
[电信官方IPv4测速](https://10000.gd.cn/#/speed)，下行能跑满2000M，CPU占用20-40%

[image]
[中科大IPv6测速](http://test6.ustc.edu.cn/)，下行能跑满2000M，CPU占用20-40%

[image]
[东北大学IPv6测速](https://speed.neu6.edu.cn/)，下行能跑满2000M，CPU占用20-40%

经搜索找到 https://github.com/openwrt/openwrt/issues/10224 ，从讨论中可知这是 OpenWrt 从 22.03 开始的 bug，在最新的 OpenWrt 23.05.2 中仍未修复（那个issue里面还在讨论，讨论中也给出了临时解决方法，但我试了不成功），不知 ImmortalWrt 这边能否进行修复呢？

### ImmortalWrt version

r27304-31bc47589e

### ImmortalWrt release

23.05.1

### ImmortalWrt target/subtarget

x86/64

### Device

J4125 i225-V(b6)*6

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

microka, 2023-11-20:
> [这个补丁](https://github.com/openwrt/openwrt/issues/10224#issuecomment-1471911970)是有效的

已经重新部署好了 ImmortalWrt 21.02.7，希望这个补丁能整合进 ImmortalWrt 下个版本。

ximiTech, 2023-12-05:
> [这个补丁](https://github.com/openwrt/openwrt/issues/10224#issuecomment-1471911970)是有效的

大佬能否给出详细的代码，例如应该在源代码的哪个目录下新增此patch文件，以及patch文件的具体内容？

skefy, 2023-12-06:
同样关注这个问题。

1715173329, 2023-12-07:
patched.

closed 2023-12-07 by 1715173329