immortalwrt/immortalwrt #1127: BFP部分不能编译

asiaon123, 2023-12-14:
### Describe the bug

```
Thu Dec 14 21:48:00 2023 kern.warn kernel: [    9.067789] failed to validate module [r8125] BTF: -22
Thu Dec 14 21:48:00 2023 kern.err kernel: [    9.073165] BPF: [110133] STRUCT
Thu Dec 14 21:48:00 2023 kern.err kernel: [    9.073847] BPF: size=8 vlen=4
Thu Dec 14 21:48:00 2023 kern.err kernel: [    9.074440] BPF:
Thu Dec 14 21:48:00 2023 kern.err kernel: [    9.075022] BPF: Invalid name
Thu Dec 14 21:48:00 2023 kern.err kernel: [    9.075591] BPF:
Thu Dec 14 21:48:00 2023 kern.warn kernel: [    9.076047] failed to validate module [r8152] BTF: -22
Thu Dec 14 21:48:00 2023 kern.info kernel: [    9.077619] usbcore: registered new device driver r8152-cfgselector
Thu Dec 14 21:48:00 2023 kern.info kernel: [    9.078702] usbcore: registered new interface driver r8152
Thu Dec 14 21:48:00 2023 kern.err kernel: [    9.090246] BPF: [110133] ENUM t
Thu Dec 14 21:48:00 2023 kern.err kernel: [    9.090905] BPF: size=4 vlen=160
Thu Dec 14 21:48:00 2023 kern.err kernel: [    9.091498] BPF:
Thu Dec 14 21:48:00 2023 kern.err kernel: [    9.091956] BPF: Invalid name
Thu Dec 14 21:48:00 2023 kern.err kernel: [    9.092521] BPF:
Thu Dec 14 21:48:00 2023 kern.warn kernel: [    9.093005] failed to validate module [r8168] BTF: -22
Thu Dec 14 21:48:00 2023 kern.err kernel: [    9.102476] BPF: [110137] ENUM P_ENDIAN_BIG
Thu Dec 14 21:48:00 2023 kern.err kernel: [    9.103234] BPF: size=4 vlen=4
Thu Dec 14 21:48:00 2023 kern.err kernel: [    9.103890] BPF:
Thu Dec 14 21:48:00 2023 kern.err kernel: [    9.104348] BPF: Invalid name
Thu Dec 14 21:48:00 2023 kern.err kernel: [    9.104917] BPF:
```

已经重新编译BPF 官方其他的都可以编译 这几个不能重新编译

### ImmortalWrt version

r28047-fa6d637512

### ImmortalWrt release

r28047-fa6d637512

### ImmortalWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Self-built image

### Steps to reproduce

默认选项 开启编译bpf

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

asiaon123, 2023-12-14:
我试了 你的编译 是没有这几个问题的 我自己编译会出现这两个不能bpf 不知道我哪里选的不对

1715173329, 2023-12-15:
btf 不支持增量编译，否则无法通过校验。
这个不影响使用，不使用 btf 可以自行关掉。

closed 2023-12-15 by 1715173329