immortalwrt/immortalwrt #1150: Failed to compile bridger module for redme AX6000 router

67113513, 2024-01-10:
### Describe the bug

touch /home/immortalwrt/build_dir/target-aarch64_cortex-a53_glibc/linux-mediatek_filogic/bpf-headers/linux-5.15.145/.prepared_3b4dfaa436dfbb20db436b7811940abe_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/immortalwrt/build_dir/target-aarch64_cortex-a53_glibc/linux-mediatek_filogic/bpf-headers/linux-5.15.145/.configured_*
rm -f /home/immortalwrt/staging_dir/target-aarch64_cortex-a53_glibc/stamp/.bpf-headers_installed
grep -vE 'CONFIG_(CPU_.*ENDIAN|HZ)' /home/immortalwrt/build_dir/target-aarch64_cortex-a53_glibc/linux-mediatek_filogic/bpf-headers/linux-5.15.145/arch/mips/configs/generic_defconfig > /home/immortalwrt/build_dir/target-aarch64_cortex-a53_glibc/linux-mediatek_filogic/bpf-headers/linux-5.15.145/.config
echo 'CONFIG_CPU_LITTLE_ENDIAN=y' >> /home/immortalwrt/build_dir/target-aarch64_cortex-a53_glibc/linux-mediatek_filogic/bpf-headers/linux-5.15.145/.config
echo 'CONFIG_CPU_MIPS64_R2=y' >> /home/immortalwrt/build_dir/target-aarch64_cortex-a53_glibc/linux-mediatek_filogic/bpf-headers/linux-5.15.145/.config
echo 'CONFIG_64BIT=y' >> /home/immortalwrt/build_dir/target-aarch64_cortex-a53_glibc/linux-mediatek_filogic/bpf-headers/linux-5.15.145/.config
grep CONFIG_HZ /home/immortalwrt/build_dir/target-aarch64_cortex-a53_glibc/linux-mediatek_filogic/linux-5.15.145/.config >> /home/immortalwrt/build_dir/target-aarch64_cortex-a53_glibc/linux-mediatek_filogic/bpf-headers/linux-5.15.145/.config
yes '' | make -C /home/immortalwrt/build_dir/target-aarch64_cortex-a53_glibc/linux-mediatek_filogic/bpf-headers/linux-5.15.145 ARCH=mips CROSS_COMPILE=mips64el-linux- LLVM=1 CC="" LD="aarch64-openwrt-linux-gnu-ld" HOSTCC="/home/immortalwrt/staging_dir/host/bin/gcc" HOSTCXX="/home/immortalwrt/staging_dir/host/bin/g++" KBUILD_HOSTLDLIBS="-L/home/immortalwrt/staging_dir/host/lib" CONFIG_SHELL="bash" INSTALL_HDR_PATH="/home/immortalwrt/build_dir/target-aarch64_cortex-a53_glibc/linux-mediatek_filogic/bpf-headers/linux-5.15.145/user_headers" oldconfig
make[3]: Entering directory '/home/immortalwrt/build_dir/target-aarch64_cortex-a53_glibc/linux-mediatek_filogic/bpf-headers/linux-5.15.145'
/bin/sh: 0: Illegal option -W
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/menu.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTLD  scripts/kconfig/conf
: unknown compiler
scripts/Kconfig.include:44: Sorry, this compiler is not supported.
make[4]: *** [scripts/kconfig/Makefile:77: oldconfig] Error 1
make[3]: *** [Makefile:628: oldconfig] Error 2
make[3]: Leaving directory '/home/immortalwrt/build_dir/target-aarch64_cortex-a53_glibc/linux-mediatek_filogic/bpf-headers/linux-5.15.145'
make[2]: *** [Makefile:122: /home/immortalwrt/build_dir/target-aarch64_cortex-a53_glibc/linux-mediatek_filogic/bpf-headers/linux-5.15.145/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 2
make[2]: Leaving directory '/home/immortalwrt/package/kernel/bpf-headers'
time: package/kernel/bpf-headers/compile#15.06#5.20#21.05
    ERROR: package/kernel/bpf-headers failed to build.
make[1]: *** [package/Makefile:128: package/kernel/bpf-headers/compile] Error 1
make[1]: Leaving directory '/home/immortalwrt'
make: *** [/home/immortalwrt/include/toplevel.mk:232: package/bridger/compile] Error 2

### ImmortalWrt version

r28559-e61b0646b5

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Redmi Router AX6000 (OpenWrt U-Boot layout)

### Image kind

Self-built image

### Steps to reproduce

make -j1 V=s package/bridger/compile

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_xiaomi_redmi-router-ax6000-ubootmod=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_AIRCRACK_NG_OPENSSL=y
# CONFIG_GDB is not set
CONFIG_LIBC="glibc"
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_CRYPTO_AUTH=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_TFTP=y
CONFIG_LIBCURL_THREADED_RESOLVER=y
CONFIG_LIBCURL_TLS_SRP=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_LIBC_USE_GLIBC=y
# CONFIG_LIBC_USE_MUSL is not set
CONFIG_MAKE_TOOLCHAIN=y
CONFIG_OPENVPN_openssl_ENABLE_FRAGMENT=y
CONFIG_OPENVPN_openssl_ENABLE_LZ4=y
CONFIG_OPENVPN_openssl_ENABLE_LZO=y
CONFIG_OPENVPN_openssl_ENABLE_PORT_SHARE=y
CONFIG_OPENVPN_openssl_ENABLE_SMALL=y
CONFIG_PACKAGE_adbyby=y
CONFIG_PACKAGE_aircrack-ng=y
CONFIG_PACKAGE_appfilter=y
CONFIG_PACKAGE_atop=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_bpftool-full=y
CONFIG_PACKAGE_bridge=y
CONFIG_PACKAGE_btop=y
CONFIG_PACKAGE_ca-certificates=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_coreutils=y
CONFIG_PACKAGE_coreutils-nohup=y
CONFIG_PACKAGE_cpulimit=y
CONFIG_PACKAGE_cpusage=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_dos2unix=y
CONFIG_PACKAGE_ethtool-full=y
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_hwinfo=y
CONFIG_PACKAGE_ip-full=y
CONFIG_PACKAGE_ipset=y
CONFIG_PACKAGE_kmod-inet-diag=y
CONFIG_PACKAGE_kmod-ipt-conntrack=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-ipset=y
CONFIG_PACKAGE_kmod-mtd-rw=y
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nf-tproxy=y
CONFIG_PACKAGE_kmod-nft-tproxy=y
CONFIG_PACKAGE_kmod-nls-base=y
CONFIG_PACKAGE_kmod-oaf=y
CONFIG_PACKAGE_kmod-pf-ring=y
CONFIG_PACKAGE_kmod-tun=y
CONFIG_PACKAGE_kmod-usb-core=y
CONFIG_PACKAGE_kmod-usb-printer=y
CONFIG_PACKAGE_libbfd=y
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libcap-bin=y
CONFIG_PACKAGE_libcap-bin-capsh-shell="/bin/sh"
CONFIG_PACKAGE_libcap-ng=y
CONFIG_PACKAGE_libcurl=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_libipset=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_liblz4=y
CONFIG_PACKAGE_liblzo=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libnghttp2=y
CONFIG_PACKAGE_libnl-core=y
CONFIG_PACKAGE_libnl-genl=y
CONFIG_PACKAGE_libopcodes=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libpcre2=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_libruby=y
CONFIG_PACKAGE_libsmartcols=y
CONFIG_PACKAGE_libstdcpp=y
CONFIG_PACKAGE_libtirpc=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libuuid=y
CONFIG_PACKAGE_libyaml=y
CONFIG_PACKAGE_lscpu=y
CONFIG_PACKAGE_lsof=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-adbyby-plus=y
CONFIG_PACKAGE_luci-app-appfilter=y
CONFIG_PACKAGE_luci-app-cpufreq=y
CONFIG_PACKAGE_luci-app-cpulimit=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-openclash=y
CONFIG_PACKAGE_luci-app-openvpn=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-app-usb-printer=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-compat=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-dashboard=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_luci-theme-material=y
CONFIG_PACKAGE_mt76-test=y
CONFIG_PACKAGE_openvpn-openssl=y
CONFIG_PACKAGE_p910nd=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ruby=y
CONFIG_PACKAGE_ruby-bigdecimal=y
CONFIG_PACKAGE_ruby-date=y
CONFIG_PACKAGE_ruby-digest=y
CONFIG_PACKAGE_ruby-enc=y
CONFIG_PACKAGE_ruby-forwardable=y
CONFIG_PACKAGE_ruby-pstore=y
CONFIG_PACKAGE_ruby-psych=y
CONFIG_PACKAGE_ruby-stringio=y
CONFIG_PACKAGE_ruby-yaml=y
CONFIG_PACKAGE_strace=y
CONFIG_PACKAGE_tcpdump=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_unzip=y
CONFIG_PACKAGE_zlib=y
CONFIG_SDK=y
CONFIG_STRIP_ARGS="--strip-all"
CONFIG_TARGET_OPTIONS=y
CONFIG_TARGET_SUFFIX="gnu"
CONFIG_USE_GLIBC=y
CONFIG_USE_STRIP=y
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-01-10:
> : unknown compiler
> scripts/Kconfig.include:44: Sorry, this compiler is not supported.

closed 2024-01-10 by 1715173329

67113513, 2024-01-10:
Fixed by chosen 'Advanced configuration options -> BPF toolchain -> Build LLVM toolchain for eBPF', should LLVM be added to bpf-headers dependency?