immortalwrt/immortalwrt #1158: R4S编译master/23.05分支总会卡在argp-standalone

kkkkllll00, 2024-01-16:
### Describe the bug

不加插件能够正常编译成功

但是加了一些插件之后，每次编译总会卡在这个位置

>
make[3]: Leaving directory '/home/admin123/immortalwrt/package/libs/libnetfilter-conntrack'
time: package/libs/libnetfilter-conntrack/compile#0.09#0.07#0.18
make[3]: Entering directory '/home/admin123/immortalwrt/package/libs/nettle'
mkdir -p /home/admin123/immortalwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
echo '8' | cmp -s - /home/admin123/immortalwrt/staging_dir/target-aarch64_generic_musl/pkginfo/libnettle.version || { echo '8' > /home/admin123/immortalwrt/staging_dir/target-aarch64_generic_musl/pkginfo/libnettle.version;  } 
SHELL= flock /home/admin123/immortalwrt/tmp/.root-copy.flock -c 'cp -fpR /home/admin123/immortalwrt/build_dir/target-aarch64_generic_musl/nettle-3.9.1/.pkgdir/libnettle/. /home/admin123/immortalwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/admin123/immortalwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.libnettle_installed
echo "libnettle" >> /home/admin123/immortalwrt/staging_dir/target-aarch64_generic_musl/pkginfo/nettle.default.install
make[3]: Leaving directory '/home/admin123/immortalwrt/package/libs/nettle'
time: package/libs/nettle/compile#0.11#0.06#0.19
make[3]: Entering directory '/home/admin123/immortalwrt/package/network/services/dnsmasq'
mkdir -p /home/admin123/immortalwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
SHELL= flock /home/admin123/immortalwrt/tmp/.root-copy.flock -c 'cp -fpR /home/admin123/immortalwrt/build_dir/target-aarch64_generic_musl/dnsmasq-full/dnsmasq-2.89/.pkgdir/dnsmasq-full/. /home/admin123/immortalwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/admin123/immortalwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.dnsmasq-full_installed
echo "dnsmasq-full" >> /home/admin123/immortalwrt/staging_dir/target-aarch64_generic_musl/pkginfo/dnsmasq.full.install
make[3]: Leaving directory '/home/admin123/immortalwrt/package/network/services/dnsmasq'
time: package/network/services/dnsmasq/full/compile#0.10#0.06#0.19
make[3]: Entering directory '/home/admin123/immortalwrt/package/utils/bzip2'
make[3]: Leaving directory '/home/admin123/immortalwrt/package/utils/bzip2'
time: package/utils/bzip2/compile#0.15#0.08#0.22
make[3]: Entering directory '/home/admin123/immortalwrt/package/libs/argp-standalone'
mkdir -p /home/admin123/immortalwrt/dl
SHELL= flock /home/admin123/immortalwrt/tmp/.argp-standalone-1.3.tar.gz.flock -c '  	/home/admin123/immortalwrt/scripts/download.pl "/home/admin123/immortalwrt/dl" "argp-standalone-1.3.tar.gz" "dec79694da1319acd2238ce95df57f3680fea2482096e483323fddf3d818d8be" "" "http://www.lysator.liu.se/~nisse/misc/"    '
‘+ curl -f --connect-timeout 20 --retry 5 --location http://www.lysator.liu.se/~nisse/misc/argp-standalone-1.3.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  560k    0  560k    0     0   1156      0 --:--:--  0:08:15 --:--:--     0’

虽然显示下载100%了，但还是这么一直卡着，最长等了一个多小时也是没反应

网络环境是有挂全局梯子的，make之前也make dowload下载了所需的dl文件

尝试本地下载好argp-standalone-1.3.tar.gz放入dl目录中，会提示HASH值不对，修改Makefile里的HASH值，又会有别的报错

### ImmortalWrt version

master和23.05

### ImmortalWrt release

master和23.05

### ImmortalWrt target/subtarget

rockchip/rk3399

### Device

Friendlywrt R4S 1G

### Image kind

Official downloaded image

### Diffconfig

```text
CONFIG_TARGET_rockchip=y
CONFIG_TARGET_rockchip_armv8=y
CONFIG_TARGET_rockchip_armv8_DEVICE_friendlyarm_nanopi-r4s=y
CONFIG_COREMARK_ENABLE_MULTITHREADING=y
CONFIG_COREMARK_NUMBER_OF_THREADS=16
CONFIG_COREMARK_OPTIMIZE_O3=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_CRYPTO_AUTH=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_TFTP=y
CONFIG_LIBCURL_THREADED_RESOLVER=y
CONFIG_LIBCURL_TLS_SRP=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_LIBSODIUM_MINIMAL=y
CONFIG_MBEDTLS_AES_C=y
CONFIG_MBEDTLS_ARMV8CE_AES_C=y
CONFIG_MBEDTLS_CMAC_C=y
CONFIG_MBEDTLS_DES_C=y
CONFIG_MBEDTLS_ECP_DP_CURVE25519_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP256K1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP256R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP384R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP521R1_ENABLED=y
CONFIG_MBEDTLS_ENTROPY_FORCE_SHA256=y
CONFIG_MBEDTLS_GCM_C=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA_ENABLED=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_PSK_ENABLED=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_RSA_ENABLED=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK_ENABLED=y
CONFIG_MBEDTLS_NIST_KW_C=y
CONFIG_MBEDTLS_RSA_NO_CRT=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_boost=y
CONFIG_PACKAGE_boost-program_options=y
CONFIG_PACKAGE_boost-system=y
CONFIG_PACKAGE_ca-certificates=y
CONFIG_PACKAGE_chinadns-ng=y
CONFIG_PACKAGE_coremark=y
CONFIG_PACKAGE_coreutils=y
CONFIG_PACKAGE_coreutils-base64=y
CONFIG_PACKAGE_coreutils-nohup=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_ddns-go=y
CONFIG_PACKAGE_ddns-scripts=y
CONFIG_PACKAGE_ddns-scripts-services=y
CONFIG_PACKAGE_dns2socks=y
CONFIG_PACKAGE_dns2tcp=y
CONFIG_PACKAGE_haproxy=y
CONFIG_PACKAGE_ip-full=y
CONFIG_PACKAGE_ipt2socks=y
CONFIG_PACKAGE_kmod-inet-diag=y
CONFIG_PACKAGE_kmod-netlink-diag=y
CONFIG_PACKAGE_kmod-nf-socket=y
CONFIG_PACKAGE_kmod-nf-tproxy=y
CONFIG_PACKAGE_kmod-nft-socket=y
CONFIG_PACKAGE_kmod-nft-tproxy=y
CONFIG_PACKAGE_kmod-tun=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libcap-bin=y
CONFIG_PACKAGE_libcap-bin-capsh-shell="/bin/sh"
CONFIG_PACKAGE_libcares=y
CONFIG_PACKAGE_libcurl=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_libev=y
CONFIG_PACKAGE_libltdl=y
CONFIG_PACKAGE_liblua5.3=y
CONFIG_PACKAGE_libmbedtls=y
CONFIG_PACKAGE_libnghttp2=y
CONFIG_PACKAGE_libpcre=y
CONFIG_PACKAGE_libpcre2=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_libruby=y
CONFIG_PACKAGE_libsodium=y
CONFIG_PACKAGE_libstdcpp=y
CONFIG_PACKAGE_libuci-lua=y
CONFIG_PACKAGE_libudns=y
CONFIG_PACKAGE_libuv=y
CONFIG_PACKAGE_libwebsockets-full=y
CONFIG_PACKAGE_libyaml=y
CONFIG_PACKAGE_luci-app-aliddns=y
CONFIG_PACKAGE_luci-app-ddns=y
CONFIG_PACKAGE_luci-app-ddns-go=y
CONFIG_PACKAGE_luci-app-openclash=y
CONFIG_PACKAGE_luci-app-passwall=y
CONFIG_PACKAGE_luci-app-shadowsocks-libev=y
CONFIG_PACKAGE_luci-app-socat=y
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_luci-i18n-aliddns-zh-cn=y
CONFIG_PACKAGE_luci-i18n-ddns-go-zh-cn=y
CONFIG_PACKAGE_luci-i18n-ddns-zh-cn=y
CONFIG_PACKAGE_luci-i18n-passwall-zh-cn=y
CONFIG_PACKAGE_luci-i18n-shadowsocks-libev-zh-cn=y
CONFIG_PACKAGE_luci-i18n-socat-zh-cn=y
CONFIG_PACKAGE_luci-i18n-ttyd-zh-cn=y
CONFIG_PACKAGE_luci-theme-argon=y
CONFIG_PACKAGE_luci-theme-openwrt=y
CONFIG_PACKAGE_microsocks=y
CONFIG_PACKAGE_openssl-util=y
CONFIG_PACKAGE_resolveip=y
CONFIG_PACKAGE_ruby=y
CONFIG_PACKAGE_ruby-bigdecimal=y
CONFIG_PACKAGE_ruby-date=y
CONFIG_PACKAGE_ruby-digest=y
CONFIG_PACKAGE_ruby-enc=y
CONFIG_PACKAGE_ruby-forwardable=y
CONFIG_PACKAGE_ruby-pstore=y
CONFIG_PACKAGE_ruby-psych=y
CONFIG_PACKAGE_ruby-stringio=y
CONFIG_PACKAGE_ruby-yaml=y
CONFIG_PACKAGE_shadowsocks-libev-config=y
CONFIG_PACKAGE_shadowsocks-libev-ss-local=y
CONFIG_PACKAGE_shadowsocks-libev-ss-redir=y
CONFIG_PACKAGE_shadowsocks-libev-ss-server=y
CONFIG_PACKAGE_shadowsocks-rust-sslocal=y
CONFIG_PACKAGE_shadowsocksr-libev-ssr-local=y
CONFIG_PACKAGE_shadowsocksr-libev-ssr-redir=y
CONFIG_PACKAGE_simple-obfs-client=y
CONFIG_PACKAGE_sing-box=y
CONFIG_PACKAGE_socat=y
CONFIG_PACKAGE_tcping=y
CONFIG_PACKAGE_transfer=y
CONFIG_PACKAGE_trojan-plus=y
CONFIG_PACKAGE_ttyd=y
CONFIG_PACKAGE_unzip=y
CONFIG_PACKAGE_v2ray-plugin=y
CONFIG_PACKAGE_xray-core=y
CONFIG_PACKAGE_zlib=y
CONFIG_SING_BOX_BUILD_CLASH_API=y
CONFIG_SING_BOX_BUILD_ECH=y
CONFIG_SING_BOX_BUILD_GVISOR=y
CONFIG_SING_BOX_BUILD_QUIC=y
CONFIG_SING_BOX_BUILD_UTLS=y
CONFIG_SING_BOX_BUILD_WIREGUARD=y
# CONFIG_TARGET_IMAGES_GZIP is not set
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-variant-release=y
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

xuexijiaoben, 2024-01-19:
请问分支master和23.05或21.02之类的分支有什么区别呀，谢谢啦

Jayucrol, 2024-01-25:
在源码dl文件夹下运行“wget http://www.lysator.liu.se/~nisse/misc/argp-standalone-1.3.tar.gz”
继续编译

closed 2024-01-25 by kkkkllll00