immortalwrt/immortalwrt #1159: "aria2c" rely on libopenssl-legacy

Shinonome28, 2024-01-16:
### Describe the bug

When start aria2c, the following error happends:
```
errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
```
The binary is from official.

After checking "libopenssl-legacy" option in config menu, this problem is fixed. 

Is there a way to remove this dependency or it is just a miss of dependency?

### ImmortalWrt version

. /etc/openwrt_release && echo $DISTRIB_REVISION

### ImmortalWrt release

23.05-SNAPSHOT

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

Lunzn FastRhino R66S

### Image kind

Self-built image

### Diffconfig

```text
CONFIG_TARGET_rockchip=y
CONFIG_TARGET_rockchip_armv8=y
CONFIG_TARGET_rockchip_armv8_DEVICE_lunzn_fastrhino-r66s=y
CONFIG_ARIA2_BITTORRENT=y
CONFIG_ARIA2_NOXML=y
CONFIG_ARIA2_OPENSSL=y
CONFIG_ARIA2_WEBSOCKET=y
CONFIG_DOCKER_CGROUP_OPTIONS=y
CONFIG_DOCKER_NET_MACVLAN=y
CONFIG_DOCKER_STO_EXT4=y
CONFIG_GNUTLS_ALPN=y
CONFIG_GNUTLS_ANON=y
CONFIG_GNUTLS_DTLS_SRTP=y
CONFIG_GNUTLS_HEARTBEAT=y
CONFIG_GNUTLS_OCSP=y
CONFIG_GNUTLS_PSK=y
CONFIG_KERNEL_CGROUP_DEVICE=y
CONFIG_KERNEL_CGROUP_FREEZER=y
CONFIG_KERNEL_CGROUP_NET_PRIO=y
CONFIG_KERNEL_EXT4_FS_POSIX_ACL=y
CONFIG_KERNEL_EXT4_FS_SECURITY=y
CONFIG_KERNEL_FS_POSIX_ACL=y
CONFIG_KERNEL_NET_CLS_CGROUP=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_CRYPTO_AUTH=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_TFTP=y
CONFIG_LIBCURL_THREADED_RESOLVER=y
CONFIG_LIBCURL_TLS_SRP=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
[... 245 lines cut ...]
CONFIG_SAMBA4_SERVER_NETBIOS=y
CONFIG_SAMBA4_SERVER_VFS=y
CONFIG_SAMBA4_SERVER_WSDD2=y
CONFIG_SING_BOX_BUILD_CLASH_API=y
CONFIG_SING_BOX_BUILD_ECH=y
CONFIG_SING_BOX_BUILD_GVISOR=y
CONFIG_SING_BOX_BUILD_QUIC=y
CONFIG_SING_BOX_BUILD_UTLS=y
CONFIG_SING_BOX_BUILD_WIREGUARD=y
CONFIG_SQLITE3_COLUMN_METADATA=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_RTREE=y
CONFIG_TARGET_ROOTFS_PARTSIZE=1024
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-variant-release=y
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-01-20:
yes, but i have no idea how to to resolve the dependency recurrence hell currently

hetaozhen, 2024-02-06:
> 是的，但我目前不知道如何解决依赖重复地狱
https://www.cnblogs.com/adam-ma/p/18003059

> ### 描述错误
> 启动aria2c时，出现以下错误：
> 
> ```
> errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
> ```
> 
> 二进制文件来自官方。
> 
> 检查配置菜单中的“libopenssl-legacy”选项后，此问题得到解决。
> 
> 有没有办法消除这种依赖性，或者只是缺少依赖性？
> 
> ### 不朽Wrt版本
> r27458-a83b379a31
> 
> ### ImmortalWrt 发布
> 23.05-快照
> 
> ### ImmortalWrt 目标/子目标
> 瑞芯微/ARMV8
> 
> ### 设备
> 朗森 FastRhino R66S
> 
> ### 图片种类
> 自建形象
> 
> ### 重现步骤
> _没有反应_
> 
> ### 实际行为
> _没有反应_
> 
> ### 预期行为
> _没有反应_
> 
> ### 附加信息
> _没有反应_
> 
> ### 差异配置
> ```
> CONFIG_TARGET_rockchip=y
> CONFIG_TARGET_rockchip_armv8=y
> CONFIG_TARGET_rockchip_armv8_DEVICE_lunzn_fastrhino-r66s=y
> CONFIG_ARIA2_BITTORRENT=y
> CONFIG_ARIA2_NOXML=y
> CONFIG_ARIA2_OPENSSL=y
> CONFIG_ARIA2_WEBSOCKET=y
> CONFIG_DOCKER_CGROUP_OPTIONS=y
> CONFIG_DOCKER_NET_MACVLAN=y
> CONFIG_DOCKER_STO_EXT4=y
> CONFIG_GNUTLS_ALPN=y
> CONFIG_GNUTLS_ANON=y
> CONFIG_GNUTLS_DTLS_SRTP=y
> CONFIG_GNUTLS_HEARTBEAT=y
> CONFIG_GNUTLS_OCSP=y
> CONFIG_GNUTLS_PSK=y
> CONFIG_KERNEL_CGROUP_DEVICE=y
> CONFIG_KERNEL_CGROUP_FREEZER=y
> CONFIG_KERNEL_CGROUP_NET_PRIO=y
> CONFIG_KERNEL_EXT4_FS_POSIX_ACL=y
> CONFIG_KERNEL_EXT4_FS_SECURITY=y
> CONFIG_KERNEL_FS_POSIX_ACL=y
> CONFIG_KERNEL_NET_CLS_CGROUP=y
> CONFIG_LIBCURL_COOKIES=y
> CONFIG_LIBCURL_CRYPTO_AUTH=y
[... 256 lines cut ...]
> CONFIG_SAMBA4_SERVER_VFS=y
> CONFIG_SAMBA4_SERVER_WSDD2=y
> CONFIG_SING_BOX_BUILD_CLASH_API=y
> CONFIG_SING_BOX_BUILD_ECH=y
> CONFIG_SING_BOX_BUILD_GVISOR=y
> CONFIG_SING_BOX_BUILD_QUIC=y
> CONFIG_SING_BOX_BUILD_UTLS=y
> CONFIG_SING_BOX_BUILD_WIREGUARD=y
> CONFIG_SQLITE3_COLUMN_METADATA=y
> CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
> CONFIG_SQLITE3_FTS3=y
> CONFIG_SQLITE3_FTS4=y
> CONFIG_SQLITE3_FTS5=y
> CONFIG_SQLITE3_RTREE=y
> CONFIG_TARGET_ROOTFS_PARTSIZE=1024
> CONFIG_boost-compile-visibility-hidden=y
> CONFIG_boost-runtime-shared=y
> CONFIG_boost-static-and-shared-libs=y
> CONFIG_boost-variant-release=y
> ```
> 
> ### 条款
> * [x] 我正在报告 ImmortalWrt 的问题，而不是不受支持的分叉。

https://www.cnblogs.com/adam-ma/p/18003059

closed 2024-03-04 by 1715173329