immortalwrt/immortalwrt #1173: Crash on NanoPC T6

xosadmin, 2024-01-26:
### Describe the bug

When booting from built-in EMMC with ImmortalWrt, it always prompts kernel panic, which is caused by sdhci.

### ImmortalWrt version

r28737-25b2312f91

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPC T6

### Image kind

Official downloaded image

### Steps to reproduce

When booting from EMMC with ImmortalWrt, there is no m.2. WiFi detected. Also, it prompts kernel panic.

### Actual behaviour

Boot log: 
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.484024] WARNING: CPU: 5 PID: 183 at drivers/mmc/host/sdhci.c:1194 sdhci_send_command+0x778/0xb84
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.484823] Modules linked in: rtw88_8822cu rtw88_8822ce rtw88_8822c rtw88_8822bu rtw88_8822be rtw88_8822b rtw88_8821cu rtw88_8821ce rtw88_8821c rtw88_8723d rtl8821ae rtl8192se rtl8192de rtl8192cu rtl8192ce rtl8192c_common rtl_usb rtl_pci qcserial pppoe ppp_async option mt7921u mt7921s mt7921e mt7921_common mt76x0u mt76x0e mt76x0_common iwlmvm iwldvm cdc_mbim btcoexist ath11k_pci ath11k ath10k_pci ath10k_core ath wireguard usb_wwan sierra_net sierra rtw88_usb rtw88_pci rtw88_core rtlwifi rtl8812au rtl8192eu rtl8189es rtl8188eu rndis_host qmi_wwan pppox ppp_mppe ppp_generic nft_fib_inet nf_flow_table_inet mt7996e mt792x_usb mt792x_lib mt7915e mt76x2u mt76x2e mt76x2_common mt76x02_usb mt76x02_lib mt7663u mt7663s mt7663_usb_sdio_common mt7615e mt7615_common mt7603e mt76_usb mt76_sdio mt76_connac_lib mt76 mhi_wwan_mbim mhi_wwan_ctrl mac80211 libchacha20poly1305 kalmia iwlwifi ipt_REJECT huawei_cdc_ncm chacha_neon cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem belkin_sa ax88179_178a asix
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.484914]  aqc111 zstd xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CLASSIFY wwan usbserial usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc rtl8150 r8152 r8125 qrtr_mhi qrtr qmi_helpers poly1305_neon nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_fullcone nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_compat nft_chain_nat nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_reject_ipv4 nf_log_syslog nf_flow_table nf_conntrack_netlink nf_conncount mhi macvlan lzo_rle lzo libcurve25519_generic libchacha kaweth ipvlan
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.492535]  iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables crc_ccitt compat cdc_wdm cdc_acm br_netfilter ax88796b ntfs3 act_connmark sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 bonding ip6_gre ip_gre gre ifb dummy ip6_udp_tunnel udp_tunnel sit ipip ip6_tunnel tunnel6 tunnel4 ip_tunnel veth tun cifs oid_registry cifs_md4 cifs_arc4 asn1_decoder dns_resolver nls_utf8 nls_cp437 wp512 twofish_generic
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.500172]  twofish_common tea serpent_generic khazad cast6_generic cast5_generic cast_common camellia_generic blowfish_generic blowfish_common anubis xts crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha256_generic libsha256 sha1_generic seqiv michael_mic md5 ecb des_generic libdes cmac authencesn authenc arc4 crypto_acompress uas uhci_hcd fsl_mph_dr_of ehci_fsl gpio_button_hotplug vfat fat exfat btrfs xor zstd_decompress zstd_compress zstd_common xor_neon raid6_pq lzo_decompress lzo_compress
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.511669] CPU: 5 PID: 183 Comm: kworker/5:1H Tainted: G        W          6.1.74 #0
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.512353] Hardware name: FriendlyElec NanoPC-T6 (DT)
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.512801] Workqueue: kblockd blk_mq_run_work_fn
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.513214] pstate: a04000c9 (NzCv daIF +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.513822] pc : sdhci_send_command+0x778/0xb84
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.514219] lr : sdhci_send_command+0x6ac/0xb84
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.514617] sp : ffff80000b0f3990
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.514907] x29: ffff80000b0f3990 x28: 0000000000000000 x27: 0000000000000002
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.515531] x26: 0000000000000000 x25: ffff000101bed800 x24: ffff000102a99e80
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.516154] x23: 000000000000000b x22: ffff000101bed540 x21: ffff000102a99ef0
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.516777] x20: ffff000102a99e80 x19: ffff000101bed540 x18: ffff80000b01d088
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.517400] x17: 7479622036393034 x16: 203a7a7328206c6c x15: 7566207369207265
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.518023] x14: 6666756220626c74 x13: ffff8000093b6de0 x12: 0000000000000bac
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.518645] x11: 00000000000003e4 x10: ffff80000951cb40 x9 : ffff8000093b6de0
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.519269] x8 : ffffffffffffffff x7 : ffff8000093b6de0 x6 : 00000001000183e4
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.519892] x5 : 0000000000000000 x4 : 0000000000000020 x3 : 0000000000000002
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.520514] x2 : fffffffffffdfbfe x1 : fffffffffffffffe x0 : 00000000ffffffe4
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.521137] Call trace:
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.521351]  sdhci_send_command+0x778/0xb84
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.521718]  sdhci_send_command_retry+0x48/0x14c
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.522122]  sdhci_request+0x74/0xd0
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.522437]  dwcmshc_request+0x3c/0x70
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.522765]  __mmc_start_request+0x70/0x1dc
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.523133]  mmc_start_request+0x88/0xb4
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.523477]  mmc_blk_mq_issue_rq+0x46c/0x8a0
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.523851]  mmc_mq_queue_rq+0x11c/0x250
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.524194]  blk_mq_dispatch_rq_list+0x19c/0x81c
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.524599]  blk_mq_do_dispatch_sched+0x298/0x310
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.525010]  __blk_mq_sched_dispatch_requests+0xfc/0x160
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.525473]  blk_mq_sched_dispatch_requests+0x3c/0x80
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.525914]  __blk_mq_run_hw_queue+0x70/0xb0
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.526289]  blk_mq_run_work_fn+0x24/0x34
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.526641]  process_one_work+0x200/0x474
Thu Jan 25 06:10:31 2024 kern.warn kernel: [  169.526993]  worker_thread+0x74/0x43c
[... 466 lines cut ...]
CONFIG_PACKAGE_ucode-mod-uloop=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_umbim=y
CONFIG_PACKAGE_unbound-daemon=y
CONFIG_PACKAGE_uqmi=y
CONFIG_PACKAGE_usb-modeswitch=y
CONFIG_PACKAGE_vlmcsd=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_wireless-regdb=y
CONFIG_PACKAGE_wpad=y
CONFIG_PACKAGE_wwan=y
CONFIG_PACKAGE_xray-core=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zerotier=y
CONFIG_PACKAGE_zlib=y
CONFIG_PARTED_READLINE=y
CONFIG_TARGET_ROOTFS_PARTSIZE=512
CONFIG_WPA_MBO_SUPPORT=y
CONFIG_WPA_MSG_MIN_PRIORITY=3
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

xosadmin, 2024-01-26:
BPF Kernel Log:
[    6.543853] BPF: 	 type_id=344 bitfield_size=0 bits_offset=840
[    6.544367] BPF:  
[    6.544544] BPF: Invalid name
[    6.544803] BPF: 
[    6.544974] failed to validate module [mac80211] BTF: -22

xosadmin, 2024-01-26:
I have tried to compile by the latest commit, but the problem of sdhci is still there when MT7922 is plugged and every driver is loaded. I tried to replace the MT7922 with QCNFA765, the sdhci crash disappeared but still turned into failed to validate module and BTF: -22 when loading all drivers.

1715173329, 2024-03-25:
fixed

closed 2024-03-25 by 1715173329