immortalwrt/immortalwrt #1177: `iptables-mod-fullconenat` 已安装但未启用

Pil0tXia, 2024-02-03:
### Describe the bug

`iptables-mod-fullconenat` opkg 包是 ImmortalWrt 默认安装的包之一，其功能是为 OpenWRT 提供 Full Cone NAT 实现。

### 环境

上游的 NAT 类型为 Full Cone NAT，upnp 开启，使用水星 D196G 路由器时，lan 中设备的 NAT 类型是 Full Cone NAT。

### 问题

在 ImmortalWrt 中，即使在 turboacc（网络 - 网络加速）中选中了`xt_FULLCONENAT`，lan 中设备的 NAT 类型仍是对称型 NAT。

[image]

[image]

“防火墙”页面中也没有看到开启完全锥型 NAT 的选项（原版 OpenWRT 的防火墙页面没有此选项，使用 fullconenat 补丁后会多出此选项）：

[image]

>图片来自 https://iyzm.net/openwrt/3313.html

### 解决方案

希望 ImmortalWrt 可以支持 Full Cone NAT！

### 相关资料

- [OpenWrt 上实现 NAT1 (Full cone NAT) 的方法，无需 DMZ/UPnP](https://www.right.com.cn/forum/thread-319827-1-1.html)
- [xt_FULLCONENAT -- Implementing RFC 3489 full cone SNAT in OpenWrt](https://forum.openwrt.org/t/xt-fullconenat-implementing-rfc-3489-full-cone-snat-in-openwrt/14816)
- [FS#1738 - Please include Full Cone NAT package · Issue #7021 · openwrt/openwrt](https://github.com/openwrt/openwrt/issues/7021)

>我使用的 ImmortalWrt 由 https://github.com/hanwckf/immortalwrt-mt798x 编译

### ImmortalWrt version

r0-e1b4487

### ImmortalWrt release

21.02-SNAPSHOT

### ImmortalWrt target/subtarget

mediatek/mt7981

### Device

Xiaomi Mi Router AX3000T

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

Hyy2001X, 2024-02-11:
试试用`firewall`而不是`firewall4`

Pil0tXia, 2024-02-18:
Refer: https://github.com/hanwckf/immortalwrt-mt798x/issues/252

closed 2024-02-18 by Pil0tXia