immortalwrt/immortalwrt #1179: cr660x build failed

liwenjie119, 2024-02-12:
### Describe the bug

sort error
and
make[5]: *** [Makefile:235: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_jcg_q20.dtb] Error 1
···
make -C /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.148 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Feb 12 13:01:54 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  CC="mipsel-openwrt-linux-musl-gcc" KERNELRELEASE=5.15.148  vmlinux vmlinuz
make[5]: Entering directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.148'
  SYNC    include/config/auto.conf.cmd
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  GEN     usr/initramfs_data.cpio
sort: write failed: 'standard output': Broken pipe
sort: write error
  LZMA    usr/initramfs_inc_data
  AS      usr/initramfs_data.o
  AR      usr/built-in.a
  CC      lib/decompress.o
  AR      lib/lib.a
···
and
···
rm -f /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_jcg_jhr-ac876m.dtb.tmp
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp  -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.148/arch/mips/boot/dts -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.148/arch/mips/boot/dts/include -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.148/include/ -undef -D__DTS__  -o /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_jcg_q20.dtb.tmp ../dts/mt7621_jcg_q20.dts
/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.148/scripts/dtc/dtc -O dtb -i../dts/ -Wno-interrupt_provider -Wno-unique_unit_address -Wno-unit_address_vs_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-simple_bus_reg    -o /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_jcg_q20.dtb /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_jcg_q20.dtb.tmp
Error: ../dts/mt7621_jcg_q20.dtsi:25.15-16 syntax error
FATAL ERROR: Unable to parse input tree
make[5]: *** [Makefile:235: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_jcg_q20.dtb] Error 1
make[5]: Leaving directory '/workdir/openwrt/target/linux/ramips/image'
make[4]: *** [Makefile:23: install] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/ramips'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/install#46.73#9.95#54.46
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:24: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:232: world] Error 2
Error: Process completed with exit code 2.
···

### ImmortalWrt version

master

### ImmortalWrt release

https://github.com/immortalwrt/immortalwrt/commit/d39f39c9ef8b752268b3af50110182e26cc69bbb

### ImmortalWrt target/subtarget

mt7621

### Device

Xiaomi Mi Router CR6609

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

xlighting2017, 2024-02-12:
it seems something get lost when merge mainline?

[image]

[image]

[image]

1715173329, 2024-02-12:
Correct. I decided to remove the pb-boot layout and add missing binding.

closed 2024-02-12 by 1715173329