immortalwrt/immortalwrt #1180: [MT7621] Poor 5GHz Performance

bagusnl, 2024-02-13:
### Describe the bug

Tested using Samsung A25 on a local [selfhosted librespeed instance](https://github.com/librespeed/speedtest), 5GHz wireless speed is abnormal, only getting 6-13Mbps download while the upload is strong at around 110Mbps. iPerf3 test also shown a lot of packet retries and connection collapsed when another instance running from the other side (practically doing `--bidir` but from 2 different processes).

<details>
  <summary>Speed Tests</summary>
  ### Librespeed on Phone
[image]

  ### iPerf3 seeing a lot of RETR
> Client : Ubuntu server connected using gigabit on USB2 (tested LAN max speed about 330Mbps)
> Server : Windows machine laptop with inbuilt Realtek GBE PCIe LAN
```sh
╭─bagusnl at server in ~ 24-02-13 - 19:14:59
╰─○ iperf3 -c 192.168.100.243 -tinf
Connecting to host 192.168.100.243, port 5201
[  5] local 192.168.100.5 port 42692 connected to 192.168.100.243 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  2.00 MBytes  16.8 Mbits/sec    2   69.3 KBytes
[  5]   1.00-2.00   sec  2.25 MBytes  18.9 Mbits/sec    0   90.5 KBytes
[  5]   2.00-3.00   sec  2.05 MBytes  17.2 Mbits/sec    0    107 KBytes
[  5]   3.00-4.00   sec  2.13 MBytes  17.8 Mbits/sec    1   89.1 KBytes
[  5]   4.00-5.00   sec  2.16 MBytes  18.1 Mbits/sec    0    106 KBytes
[  5]   5.00-6.00   sec  1.67 MBytes  14.0 Mbits/sec   19   79.2 KBytes
[  5]   6.00-7.00   sec  2.14 MBytes  18.0 Mbits/sec    1   96.2 KBytes
[  5]   7.00-8.00   sec  1.91 MBytes  16.1 Mbits/sec    0    110 KBytes
[  5]   8.00-9.00   sec   809 KBytes  6.63 Mbits/sec   19   55.1 KBytes
# Started another client instance from the Windows side
[  5]   9.00-10.01  sec  0.00 Bytes  0.00 bits/sec   33   2.83 KBytes
[  5]  10.01-11.00  sec   233 KBytes  1.92 Mbits/sec   25   9.90 KBytes
[  5]  11.00-12.00  sec  0.00 Bytes  0.00 bits/sec    8   15.6 KBytes
[  5]  12.00-13.00  sec   233 KBytes  1.91 Mbits/sec   13   11.3 KBytes
[  5]  13.00-14.00  sec  0.00 Bytes  0.00 bits/sec    9   5.66 KBytes
[  5]  14.00-15.00  sec  0.00 Bytes  0.00 bits/sec    5   4.24 KBytes
[  5]  15.00-16.01  sec   233 KBytes  1.90 Mbits/sec    2   12.7 KBytes
[  5]  16.01-17.00  sec  0.00 Bytes  0.00 bits/sec   15   5.66 KBytes
[  5]  17.00-18.00  sec  0.00 Bytes  0.00 bits/sec    5   7.07 KBytes
[  5]  18.00-19.00  sec  0.00 Bytes  0.00 bits/sec    6   4.24 KBytes
[  5]  19.00-20.00  sec  0.00 Bytes  0.00 bits/sec    1   4.24 KBytes
[  5]  20.00-21.00  sec  0.00 Bytes  0.00 bits/sec    5   4.24 KBytes
[  5]  21.00-22.00  sec  0.00 Bytes  0.00 bits/sec    5   4.24 KBytes
^C[  5]  22.00-22.60  sec   933 KBytes  12.8 Mbits/sec    3   4.24 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-22.60  sec  18.7 MBytes  6.94 Mbits/sec  177             sender
[  5]   0.00-22.60  sec  0.00 Bytes  0.00 bits/sec                  receiver
```
</details>

### ImmortalWrt version

r27304-31bc47589e

### ImmortalWrt release

23.05.1

### ImmortalWrt target/subtarget

ramips/mt7621

### Device

D-Link DIR-878 A1

### Image kind

Official downloaded image

### Steps to reproduce

1. Connect client A with LAN and client B with 5GHz to the router
2. Start iPerf3 server on client B, run iPerf3 as client on client A
3. Observe `Retr` values
4. While the other running, start iPerf3 server on client A and test from client B
5. Observe one of the speed plummeted to 0

### Actual behaviour

A lot of packet losses and connection dropouts

### Expected behaviour

Stable connection

### Additional info

<details>
  <summary>Wireless Config</summary>
config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option channel 'auto'
        option band '2g'
        option htmode 'HT20'
        option country 'AU'
        option cell_density '2'
        option mu_beamformer '1'
        option txpower '28'
        option vendor_vht '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid *******
        option encryption 'psk2+ccmp'
        option key *******
        option ieee80211w '0'
        option ifname 'ap-2.4g'
        option dtim_period '5'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option channel 'auto'
        option band '5g'
        option htmode 'VHT40'
        option country 'AU'
        option disabled '0'
        option cell_density '2'
        option mu_beamformer '1'
        option txpower '24'
        option beacon_int '500'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid *******
        option encryption 'psk2+ccmp'
        option ifname 'ap-5g'
        option key *******
        option wnm_sleep_mode '1'
        option dtim_period '1'
        option skip_inactivity_poll '1'
        option disassoc_low_ack '0'
        option wpa_group_rekey '86400'
</details>

<details>
  <summary>Network Configs</summary>
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fdc4:09ae:8179::/48'
        option packet_steering '1'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.100.1'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option gateway '192.168.1.1'
        list dns '45.90.28.72'
        list dns '45.90.30.72'

config interface 'wan'
        option device 'wan'
        option proto 'static'
        option ipaddr '192.168.1.2'
        option netmask '255.255.255.0'
        option gateway '192.168.1.1'
        list dns '45.90.30.72'
        list dns '45.90.28.72'

config interface 'wan6'
        option device 'wan'
        option proto 'dhcpv6'
        option reqaddress 'try'
        option reqprefix 'auto'

</details>

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

scropion86, 2024-03-03:
since this hardware is commonly have this issue, please change the fragmentation as on this post and you should be OK
[https://forum.openwrt.org/t/question-xiaomi-mi-r3g-mir3g-5ghz-relayd-19-07-upload-performance/50248/16](url)

bagusnl, 2024-03-03:
> since this hardware is commonly have this issue, please change the fragmentation as on this post and you should be OK [https://forum.openwrt.org/t/question-xiaomi-mi-r3g-mir3g-5ghz-relayd-19-07-upload-performance/50248/16](url)

I already have both fragmentation off, but it doesn't seem to help.
I'll try to reinstall the router with snapshot and see how it goes

f0re1gnKey, 2024-04-24:
> https://forum.openwrt.org/t/question-xiaomi-mi-r3g-mir3g-5ghz-relayd-19-07-upload-performance/50248/16

Does snapshot behave better?