immortalwrt/immortalwrt #1185: immortalwrt23.05.1实时信息中的连接无法查看

LemonCrab666, 2024-02-17:
### Describe the bug

23.05.1 x86实时信息中的连接打开会加载然后就显示登录过期，连接数在九千多
虚拟机中测试连接数二十多无此问题，master也是如此，系统日志中无报错
[image]
[image]
[image]

### ImmortalWrt version

r27304-31bc47589e

### ImmortalWrt release

23.05.1

### ImmortalWrt target/subtarget

x86/64

### Device

CncTion J4125-4L

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

science2468, 2024-03-22:
截图看看概览和日志，顺便把主题换成默认的。

microka, 2024-04-02:
我也试过，连接数一多起来去看实时信息里的连接，看着看着会卡住，过一会儿要重新登录。

microka, 2024-05-02:
[image]
[image]
[image]
[image]
ImmortalWrt 23.05.2，连接数 10000+，实时信息连接情况无法查看。

microka, 2024-05-02:
[image]
现在面板还进不去了

liuweiGL, 2024-06-28:
需要我提供什么信息给你们调试吗

aYangQT, 2024-08-13:
一样的问题，问过大佬了，连接数过大，实时信息-连接会卡死

ternam, 2024-10-21:
Same issue here on 23.05 version.

closed 2024-12-12 by LemonCrab666

microka, 2024-12-13:
@LemonCrab666 请问这个问题解决了吗？