immortalwrt/immortalwrt #1204: miniupnpd中规则无法自动清除

kxjhcmc, 2024-03-05:
### Describe the bug

miniupnpd中规则无法自动清除，设置了清除清除规则阈值为1，定时清除规则 1800秒，依然不会自动清除无效规则

### ImmortalWrt version

r29272-f08b89f121

### ImmortalWrt release

 SNAPSHOT

### ImmortalWrt target/subtarget

x86/64

### Device

CncTion J4125-4L-I225

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-03-05:
https://github.com/immortalwrt/immortalwrt/discussions/1190
垃圾 upnp 免问即可。

closed 2024-03-05 by 1715173329

kxjhcmc, 2024-09-30:
没有解决，估计国外的软件都比较友好，会自己删除自己映射的端口，不像国内软件那么流氓，不断映射新端口，不删除老端口