immortalwrt/immortalwrt #1205: 最新代码编译后web页面访问报错

hepengshuai, 2024-03-16:
### Describe the bug

left-hand side expression is null
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 133, byte 10:
  called from function [arrow function] (/usr/share/ucode/luci/runtime.uc:141:63)
  called from function render ([C])
  called from function [anonymous function] (/usr/share/ucode/luci/runtime.uc:141:64)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:786:47)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:1003:48)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `            die(ex);`
  Near here --------^

### ImmortalWrt version

master

### ImmortalWrt release

master

### ImmortalWrt target/subtarget

x86

### Device

pve

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-03-19:
should be fixed in recent version

closed 2024-03-19 by 1715173329

xiaol1n7, 2024-10-12:
> should be fixed in recent version

一摸一样的报错。。23.05.04还是出现了，x86-64