immortalwrt/immortalwrt #1213: qca-ssdk-shell 编译错误

liyiranlab, 2024-03-26:
### Describe the bug

      | ^~~~~~~~~~~~
In file included from fal_port_ctrl.c:21:
/home/ran/AX6/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/qca-ssdk-shell-2022.09.12~b43fba2d/include/fal/fal_port_ctrl.h:599:5: note: previous declaration of 'fal_port_cdt' with type 'sw_error_t(a_uint32_t,  fal_port_t,  a_uint32_t,  fal_cable_status_t *, a_uint32_t *)' {aka 'sw_error_t(unsigned int,  unsigned int,  unsigned int,  fal_cable_status_t *, unsigned int *)'}
  599 |     fal_port_cdt(a_uint32_t dev_id, fal_port_t port_id, a_uint32_t mdi_pair,
      |     ^~~~~~~~~~~~
cc1: all warnings being treated as errors
make[5]: *** [/home/ran/AX6/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/qca-ssdk-shell-2022.09.12~b43fba2d/make/target.mk:13: fal_port_ctrl.o] Error 1
make[5]: Leaving directory '/home/ran/AX6/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/qca-ssdk-shell-2022.09.12~b43fba2d/src/fal_uk'
make[4]: *** [Makefile:28: uslib_o] Error 1
make[4]: Leaving directory '/home/ran/AX6/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/qca-ssdk-shell-2022.09.12~b43fba2d'
make[3]: *** [Makefile:51: /home/ran/AX6/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/qca-ssdk-shell-2022.09.12~b43fba2d/.built] Error 2
make[3]: Leaving directory '/home/ran/AX6/immortalwrt/package/network/utils/qca-ssdk-shell'
time: package/network/utils/qca-ssdk-shell/compile#3.15#1.20#6.84
    ERROR: package/network/utils/qca-ssdk-shell failed to build.
make[2]: *** [package/Makefile:129: package/network/utils/qca-ssdk-shell/compile] Error 1
make[2]: Leaving directory '/home/ran/AX6/immortalwrt'
make[1]: *** [package/Makefile:123: /home/ran/AX6/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ran/AX6/immortalwrt'
make: *** [/home/ran/AX6/immortalwrt/include/toplevel.mk:233: world] Error 2

### ImmortalWrt version

158e54c  

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

AX6

### Device

Redmi AX6 (stock layout)

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

mitsukileung, 2024-03-26:
> 目前仓库有这个包吗？如果是其他地方搞过来的，要找包的版本是什么，我编译没有问题。。。。

https://github.com/immortalwrt/immortalwrt/tree/master/package/network/utils/qca-ssdk-shell

1715173329, 2024-03-27:
removed