immortalwrt/immortalwrt #1234: OrangePi Zero3 CPU主频问题

ZZM365-CN, 2024-04-08:
### Describe the bug

用immortalwrt master分支编译OrangePi Zero3，添加luci-app-cpufreq后发现Zero3CPU最高主频只能调到1416000。看了下官网给的信息，CPU的主频是可以调到1.5GHz的。看大佬能不能修一下。
[image]

### ImmortalWrt version

r29831-2e9cc64cec

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

sunxi/cortexa53

### Device

OrangePi Zero3

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-04-08:
根据芯片 id 决定频率上限。

closed 2024-04-08 by 1715173329

ZZM365-CN, 2024-04-08:
忘记说芯片参数了，lede那好像可以到1.5GHz的
[image]