immortalwrt/immortalwrt #1245: Request New Hardware Support: Rockchip | Add Rongpin King3399 support

telanxu, 2024-04-12:
### Describe the bug

Request New Hardware Support

Target: Rockchip
Sub-Target: RK3399

dts come from https://github.com/HIWLYF/build-linux/blob/master/boot/dts/rockchip/mainline/rk3399-rp-king3399.dts

Thanks a lot.

### ImmortalWrt version

master

### ImmortalWrt release

23.05.2

### ImmortalWrt target/subtarget

Rockchip/RK3399

### Device

Unsupported: Rongpin King 3399

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-04-12:
PR is welcome.

telanxu, 2024-04-12:
Sorry, I am not a coder, I can not create a workable source code for rongpin king3399, so, no PR for your reference... :-(