immortalwrt/immortalwrt #1273: 更改2.4G Wi-Fi 通道宽度导致`ksoftirqd` CPU 占用高的问题

lambdaplus, 2024-04-25:
### Describe the bug

[image]
把图片中的通道宽度改成40MHZ后，导致 `[ksoftirqd/0]` 和 `[ksoftirqd/1]` CPU占用高，合计达到 81%。当改成默认值 20MHZ 后，二者CPU占用立马降到 0 ，整体CPU占用也降下来了 ，温度也降下来了
[image]

昨天怀疑是 `wireguard` 的问题，把其关掉后 CPU 占用暂时解决了，今天发现 CPU 占用又高了，用上述方式解决了，后续继续观察。

### ImmortalWrt version

r27625-416c8c5c91

### ImmortalWrt release

23.05.2

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Mi Router AX3000T

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

lambdaplus, 2024-04-29:
几天用下来都挺好，CPU占用也正常了，可以确定是上述所说的问题了

lambdaplus, 2024-04-30:
可能是我哪里姿势不对吧，继续观察

lambdaplus, 2024-04-30:
刚突然发现一个现象：
我明明已经把 `2.4G` 无线的通道宽度由`20MHz`改成`40MHz`了
[image]

可是，在 状态-> 信道分析 却发现 信道宽度 依然是 `20MHz`
[image]

5G的无线是正常的。

lambdaplus, 2024-05-17:
在  sysupgrade 里重刷了一遍固件，CPU占用正常了。看来从原厂固件刷 immortalwrt ，最好是刷完 factory 固件后再刷两遍 sysupgrade 固件。也有一种可能是我刷了一遍 sysupgrade 固件后没有重启导致的？

closed 2024-05-17 by lambdaplus

SuperKali, 2024-10-31:
Still present on 23.05.4, today for the first time having made the switch from openwrt to immortalwrt I encountered the same problem

ccandjay, 2025-11-24:
在23.05.4版本，我也遇到同样的问题，每次将2.4g重启才可以