immortalwrt/immortalwrt #1280: Orange Pi 5 Plus build don't boot

mplaviak, 2024-05-01:
### Describe the bug

I'm trying to use snapshot compilation on my Orange Pi 5 Plus. but it is not booting, I tried using the memory card, and the nvme SSD

### ImmortalWrt version

snapshot

### ImmortalWrt release

snapshot

### ImmortalWrt target/subtarget

armv8

### Device

Orange Pi 5 Plus

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-05-01:
no log = no solution

mplaviak, 2024-05-01:
> no log = no solution

How do I get a log? the device does not boot, and on the computer it appears as an unrecognized device. The compilation is for the sd card or emmc, correct?

1715173329, 2024-05-01:
It's a development board, so of course, from serial.

mplaviak, 2024-05-01:
I will buy a serial cable, and send the log here

Y-ZD-heheka, 2024-05-05:
the same problem ， looking for solution

Y-ZD-heheka, 2024-05-05:
After one update of U-Boot, it could not start up anymore.

MashiroYae, 2024-05-05:
same as NanoPi 6 https://github.com/immortalwrt/immortalwrt/issues/1285

mplaviak, 2024-05-06:
I believe the problem its because the kernel 6.6. When I test the compilation with the Kernel 6.1, the system works correctly.

Y-ZD-heheka, 2024-05-17:
chuange boot device to emmc

1715173329, 2024-08-05:
by checking logs in #1285 I found the sdmmc seems broken... under investigating

1715173329, 2024-08-05:
try a3a117453659a8fa898b931c951b24c1980771b2

closed 2024-09-02 by 1715173329