immortalwrt/immortalwrt #1285: NanoPi R6S 不能启动

MashiroYae, 2024-05-04:
### Describe the bug

```
U-Boot SPL 2024.04-ImmortalWrt-r30003-4f4e8820bd (Apr 23 2024 - 07:33:34 +0000)
Trying to boot from MMC1
INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-682-g4ca8a8422:derrick.huang, fwver: v1.45
NOTICE:  BL31: Built : 10:11:21, Dec 27 2023
INFO:    spec: 0x13
INFO:    code: 0x88
INFO:    ext 32k is not valid
INFO:    ddr: stride-en 4CH
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    valid_cpu_msk=0xff bcore0_rst = 0x0, bcore1_rst = 0x0
INFO:    l3 cache partition cfg-0
INFO:    system boots from cpu-hwid-0
INFO:    disable memory repair
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
INFO:    dfs DDR fsp_params[0].freq_mhz= 2112MHz
INFO:    dfs DDR fsp_params[1].freq_mhz= 528MHz
INFO:    dfs DDR fsp_params[2].freq_mhz= 1068MHz
INFO:    dfs DDR fsp_params[3].freq_mhz= 1560MHz
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0xa00000
INFO:    SPSR = 0x3c9

U-Boot 2024.04-ImmortalWrt-r30003-4f4e8820bd (Apr 23 2024 - 07:33:34 +0000)

Model: FriendlyElec NanoPi R6S
DRAM:  8 GiB (effective 7.7 GiB)
Core:  315 devices, 24 uclasses, devicetree: separate
MMC:   mmc@fe2c0000: 0, mmc@fe2e0000: 1
Loading Environment from nowhere... OK
In:    serial@feb50000
Out:   serial@feb50000
Err:   serial@feb50000
Model: FriendlyElec NanoPi R6S
Net:   eth0: ethernet@fe1c0000
Hit any key to stop autoboot:  0 
Scanning for bootflows in all bootdevs
Seq  Method       State   Uclass    Part  Name                      Filename
---  -----------  ------  --------  ----  ------------------------  ----------------
Scanning global bootmeth 'efi_mgr':
No EFI system partition
No EFI system partition
Failed to persist EFI variables
No EFI system partition
Failed to persist EFI variables
No EFI system partition
Failed to persist EFI variables
No EFI system partition
Failed to persist EFI variables
No EFI system partition
Failed to persist EFI variables
  0  efi_mgr      ready   (none)       0  <NULL>                    
** Booting bootflow '<NULL>' with efi_mgr
Loading Boot0000 'mmc 0' failed
Loading Boot0001 'mmc 1' failed
EFI boot manager: Cannot load any image
Boot failed (err=-14)
Scanning bootdev 'mmc@fe2e0000.bootdev':
Scanning bootdev 'mmc@fe2c0000.bootdev':
  1  script       ready   mmc          1  mmc@fe2c0000.bootdev.part /boot.scr
** Booting bootflow 'mmc@fe2c0000.bootdev.part_1' with script
6210936 bytes read in 519 ms (11.4 MiB/s)
[... 404 lines cut ...]
```

### ImmortalWrt version

snapshots

### ImmortalWrt release

snapshots

### ImmortalWrt target/subtarget

 rockchip/armv8

### Device

NanoPi R6S 

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-05-06:
I cannot reproduce this issue on recent snapshots build.

MashiroYae, 2024-05-06:
> U-Boot 2024.04-ImmortalWrt-r30213-db65054a17 (May 06 2024 - 04:02:24 +0000)

```
DDR 9fffbe1e78 cym 24/02/04-10:09:20,fwver: v1.16
LPDDR4X, 2112MHz
channel[0] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[1] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[2] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[3] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
Manufacturer ID:0x1
CH0 RX Vref:28.8%, TX Vref:18.8%,18.8%
CH1 RX Vref:25.8%, TX Vref:21.8%,21.8%
CH2 RX Vref:25.0%, TX Vref:20.8%,20.8%
CH3 RX Vref:27.1%, TX Vref:20.8%,19.8%
change to F1: 528MHz
change to F2: 1068MHz
change to F3: 1560MHz
change to F0: 2112MHz
out

U-Boot SPL 2024.04-ImmortalWrt-r30213-db65054a17 (May 06 2024 - 04:02:24 +0000)
Trying to boot from MMC1
INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-682-g4ca8a8422:derrick.huang, fwver: v1.45
NOTICE:  BL31: Built : 10:11:21, Dec 27 2023
INFO:    spec: 0x13
INFO:    code: 0x88
INFO:    ext 32k is valid
INFO:    ddr: stride-en 4CH
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    valid_cpu_msk=0xff bcore0_rst = 0x0, bcore1_rst = 0x0
INFO:    l3 cache partition cfg-0
INFO:    system boots from cpu-hwid-0
INFO:    disable memory repair
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
INFO:    dfs DDR fsp_params[0].freq_mhz= 2112MHz
INFO:    dfs DDR fsp_params[1].freq_mhz= 528MHz
INFO:    dfs DDR fsp_params[2].freq_mhz= 1068MHz
INFO:    dfs DDR fsp_params[3].freq_mhz= 1560MHz
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0xa00000
INFO:    SPSR = 0x3c9

U-Boot 2024.04-ImmortalWrt-r30213-db65054a17 (May 06 2024 - 04:02:24 +0000)

Model: FriendlyElec NanoPi R6S
DRAM:  8 GiB (effective 7.7 GiB)
Core:  315 devices, 24 uclasses, devicetree: separate
MMC:   mmc@fe2c0000: 0, mmc@fe2e0000: 1
Loading Environment from nowhere... OK
In:    serial@feb50000
Out:   serial@feb50000
Err:   serial@feb50000
Model: FriendlyElec NanoPi R6S
Net:   eth0: ethernet@fe1c0000
Hit any key to stop autoboot:  0 
Scanning for bootflows in all bootdevs
Seq  Method       State   Uclass    Part  Name                      Filename
---  -----------  ------  --------  ----  ------------------------  ----------------
Scanning global bootmeth 'efi_mgr':
No EFI system partition
No EFI system partition
Failed to persist EFI variables
No EFI system partition
Failed to persist EFI variables
[... 410 lines cut ...]
[    0.985594] pci 0004:40:00.0: PME# supported from D0 D1 D3hot
[    0.988242] pci 0004:40:00.0: Primary bus is hard wired to 0
[    0.988736] pci 0004:40:00.0: bridge configuration invalid ([bus 01-ff]), reconfiguring
[    0.989542] pci 0004:41:00.0: [10ec:8125] type 00 class 0x020000
[    0.990111] pci 0004:41:00.0: reg 0x10: [io  0x0000-0x00ff]
[    0.990655] pci 0004:41:00.0: reg 0x18: [mem 0x00000000-0x0000ffff 64bit]
[    0.991282] pci 0004:41:00.0: reg 0x20: [mem 0x00000000-0x00003fff 64bit]
[    0.992205] pci 0004:41:00.0: supports D1 D2
[    0.992579] pci 0004:41:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[    1.003971] pci_bus 0004:41: busn_res: [bus 41-4f] end is updated to 41
[    1.004563] pci 0004:40:00.0: BAR 8: assigned [mem 0xf4200000-0xf42fffff]
[    1.005159] pci 0004:40:00.0: BAR 6: assigned [mem 0xf4300000-0xf430ffff pref]
[    1.005789] pci 0004:40:00.0: BAR 7: assigned [io  0x100000-0x100fff]
[    1.006353] pci 0004:41:00.0: BAR 2: assigned [mem 0xf4200000-0xf420ffff 64bit]
[    1.007018] pci 0004:41:00.0: BAR 4: assigned [mem 0xf4210000-0xf4213fff 64bit]
[    1.007684] pci 0004:41:00.0: BAR 0: assigned [io  0x100000-0x1000ff]
[    1.008266] pci 0004:40:00.0: PCI bridge to [bus 41]
[    1.008700] pci 0004:40:00.0:   bridge window [io  0x100000-0x100fff]
[    1.009264] pci 0004:40:00.0:   bridge window [mem 0xf4200000-0xf42fffff]
[    1.011431] pcieport 0004:40:00.0: PME: Signaling with IRQ 95
[    1.012410] pcieport 0004:40:00.0: AER: enabled with IRQ 96
[    1.013952] clk: Disabling unused clocks
[    1.015483] dw-apb-uart feb50000.serial: forbid DMA for kernel console
[    1.016104] Waiting for root device PARTUUID=5452574f-02...
```

MashiroYae, 2024-05-07:
试过3张不同的TF卡
在 Windows 下有用 rufus win32diskimager
Linux 下直接 dd

Y-ZD-heheka, 2024-05-07:
https://github.com/immortalwrt/immortalwrt/issues/1280#issuecomment-2095109285

Y-ZD-heheka, 2024-05-07:
https://github.com/openwrt/openwrt/issues/13361#issuecomment-1695642452，may help

Y-ZD-heheka, 2024-05-07:
https://github.com/openwrt/openwrt/issues/13361#issuecomment-1695642452
may help

1715173329, 2024-05-07:
> 试过3张不同的TF卡 在 Windows 下有用 rufus win32diskimager Linux 下直接 dd

mmc@fe2e0000 指示此设备目前从 eMMC 启动。
尝试在 maskrom 模式下写入解压后的镜像。

MashiroYae, 2024-05-07:
活了！
那 方便看看能不能修修 TF 卡启动？

1715173329, 2024-05-07:
把 eMMC 清空就能启动了，主线目前强制优先 emmc

closed 2024-05-07 by MashiroYae

MashiroYae, 2024-05-07:
我之前也是清空的状态
但是就卡那边了

Y-ZD-heheka, 2024-05-07:
> 把 eMMC 清空就能启动了，主线目前强制优先 emmc

没买emmc 😂糙