immortalwrt/immortalwrt #1301: freeswitch软件包提示libpcre不存在

VIKINGYFY, 2024-05-18:
### Describe the bug

云编译X86 MTK RK系列均会报
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libpcre', which does not exist

### ImmortalWrt version

ImmortalWrt SNAPSHOT r0-f80be24

### ImmortalWrt release

LuCI Master 24.138.36292~0395a75

### ImmortalWrt target/subtarget

ImmortalWrt SNAPSHOT r0-f80be24

### Device

X86 MTK RK

### Image kind

Self-built image

### Additional info

安装的依赖列表：
ack antlr3 asciidoc autoconf automake autopoint binutils bison build-essential bzip2 ccache cmake cpio curl device-tree-compiler dos2unix ecj fakeroot fastjar flex g++-multilib gawk gcc-multilib genisoimage gettext git gnutls-dev gperf haveged help2man intltool jq lib32gcc-s1 libc6-dev-i386 libelf-dev libglib2.0-dev libgmp3-dev libltdl-dev libmpc-dev libmpfr-dev libncurses5-dev libncursesw5 libncursesw5-dev libpython3-dev libreadline-dev libssl-dev libtool libyaml-dev libz-dev lld llvm lrzsz mkisofs msmtp nano ninja-build p7zip p7zip-full patch pkgconf python2 python3 python3-docutils python3-pip python3-ply python3-pyelftools qemu-utils quilt re2c rsync scons squashfs-tools subversion swig texinfo uglifyjs unzip upx-ucl vim wget xmlto xxd zlib1g-dev zstd

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

closed 2024-05-18 by 1715173329

jt8823, 2024-05-31:
同样问题

iiQi, 2024-06-03:
同样问题

RickyCong, 2024-08-13:
我的是友善r6s rk3588s报错libpcre没有