immortalwrt/immortalwrt #1326: nokia_ea0326gmp 引导文件错误

cyh92, 2024-06-09:
### Describe the bug

Using ethernet@15100000 device
TFTP from server 192.168.1.254; our IP address is 192.168.1.1
Filename 'immortalwrt-mediatek-filogic-nokia_ea0326gmp-preloader.bin'.
ERROR: reserving fdt memory region failed (addr=4fc00000 size=100000 flags=4)
Load address: 0x46000000

### ImmortalWrt version

R22.03

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

Nokia EA0326GMP

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-06-09:
> Steps to reproduce
>
> No response

1715173329, 2024-06-25:
Since there's no enough information for debugging, I will close it for now.

If you want to update bl2, you can just do it in OpenWrt.
```sh
opkg update && opkg install kmod-mtd-rw
insmod mtd-rw i_want_a_brick=1
mtd write immortalwrt-mediatek-filogic-nokia_ea0326gmp-preloader.bin bl2
```

closed 2024-06-25 by 1715173329