immortalwrt/immortalwrt #1329: OrangePi Zero2 stucks at starting kernel

lyxikun, 2024-06-12:
### Describe the bug

Firstly I flashed Openwrt firmware from firmware selector and it works, then I downloaded image from Immoralwrt firmware selector and it stucks. So  I tried to add 6.6 kernel patches for h616 one by one and found out "016-v6.9-arm64-dts-allwinner-h616-Add-thermal-sensor-and-zones.patch" leads to this problem.

### ImmortalWrt version

r30650-6de93879c1

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

sunxi/cortexa53

### Device

OrangePi Zero2

### Image kind

Self-built image

### Steps to reproduce

Just set target to OrangePi Zero2 in menuconfig and change nothing, then compile and flash into the SD card.

### Actual behaviour

Booting into system.

### Expected behaviour

Stuck at booting kernel

### Additional info

 U-Boot 2024.01-ImmortalWrt-r30627-ac70334e80 (Jun 07 2024 - 08:39:12 +0000) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: OrangePi Zero2
DRAM:  512 MiB
Core:  54 devices, 22 uclasses, devicetree: separate
WDT:   Not starting watchdog@30090a0
MMC:   mmc@4020000: 0
Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1... 
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Allwinner mUSB OTG (Peripheral)
Net:   eth0: ethernet@5020000using musb-hdrc, OUT ep1out IN ep1in STATUS ep2in
MAC de:ad:be:ef:00:01
HOST MAC de:ad:be:ef:00:00
RNDIS ready
, eth1: usb_ether
starting USB...
Bus usb@5200000: USB EHCI 1.00
Bus usb@5200400: USB OHCI 1.0
scanning bus usb@5200000 for devices... 1 USB Device(s) found
scanning bus usb@5200400 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
467 bytes read in 1 ms (456.1 KiB/s)
## Executing script at 4fc00000
19931144 bytes read in 828 ms (23 MiB/s)
18623 bytes read in 3 ms (5.9 MiB/s)
Moving Image from 0x40080000 to 0x40200000, end=41560000
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
Working FDT set to 4fa00000
   Loading Device Tree to 0000000049ff8000, end 0000000049fff8be ... OK
Working FDT set to 49ff8000

Starting kernel ...

### Diffconfig

```text
CONFIG_TARGET_sunxi=y
CONFIG_TARGET_sunxi_cortexa53=y
CONFIG_TARGET_sunxi_cortexa53_DEVICE_xunlong_orangepi-zero2=y
CONFIG_SUNXI_SD_BOOT_PARTSIZE=256
CONFIG_TARGET_ROOTFS_PARTSIZE=1024
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-06-25:
Sorry I missed your issue. It sounds a little bit weird as this patch also exists in upstream.
I will check if the same issue happens on Zero3.

1715173329, 2024-06-27:
should work now.

closed 2024-06-28 by lyxikun