immortalwrt/immortalwrt #1331: ERROR: package/feeds/small/sing-box failed to build.

netlovehf, 2024-06-12:
### Describe the bug

touch /home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3/.configured_78a76db87670267a8272860ec96b54c3
rm -f /home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3/.built
touch /home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/hzybb/immortalwrt/tmp/go-build" GO_MOD_CACHE_DIR="/home/hzybb/immortalwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/sagernet/sing-box" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/sagernet/sing-box/cmd/sing-box" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3" GO_BUILD_DIR="/home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3/.go_work/build" GO_BUILD_BIN_DIR="/home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3/.go_work/build/bin/linux_arm64" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/hzybb/immortalwrt/staging_dir/target-aarch64_generic_musl/usr/share/gocode/src" GOOS="linux" GOARCH="arm64" GO386="" GOAMD64="" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3=sing-box-1.9.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/include -I/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/fortify -I/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include" CGO_CXXFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3=sing-box-1.9.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/lib -L/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib -znow -zrelro" CGO_ENABLED=1 GOPATH="/home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3/.go_work/build" GOCACHE="/home/hzybb/immortalwrt/tmp/go-build" GOMODCACHE="/home/hzybb/immortalwrt/dl/go-mod-cache" GOENV=off /usr/bin/env bash /home/hzybb/immortalwrt/feeds/packages/lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1678323167' -linkmode external -extldflags '-L/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/lib -L/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags " -X github.com/sagernet/sing-box/constant.Version=v1.9.3 -buildid '1678323167' -linkmode external -extldflags '-L/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/lib -L/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'" -tags "with_clash_api,with_dhcp,with_ech,with_gvisor,with_quic,with_utls,with_wireguard" 
Finding targets

Building targets
../../common/tls/reality_client.go:10:2: package crypto/ecdh is not in GOROOT (/home/hzybb/immortalwrt/staging_dir/hostpkg/lib/go-cross/src/crypto/ecdh)

make[3]: *** [Makefile:153: /home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3/.built] Error 1
make[3]: Leaving directory '/home/hzybb/immortalwrt/feeds/small/sing-box'
time: package/feeds/small/sing-box/compile#3.18#2.50#4.28
    ERROR: package/feeds/small/sing-box failed to build.
make[2]: *** [package/Makefile:116: package/feeds/small/sing-box/compile] Error 1
make[2]: Leaving directory '/home/hzybb/immortalwrt'
make[1]: *** [package/Makefile:110: /home/hzybb/immortalwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/hzybb/immortalwrt'
make: *** [/home/hzybb/immortalwrt/include/toplevel.mk:230: world] Error 2

### ImmortalWrt version

openwrt-18.06-k5.4

### ImmortalWrt release

sing-box-1.9.3

### ImmortalWrt target/subtarget

ERROR: package/feeds/small/sing-box failed to build.

### Device

R4S

### Image kind

Official downloaded image

### Steps to reproduce

touch /home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3/.configured_78a76db87670267a8272860ec96b54c3
rm -f /home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3/.built
touch /home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/hzybb/immortalwrt/tmp/go-build" GO_MOD_CACHE_DIR="/home/hzybb/immortalwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/sagernet/sing-box" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/sagernet/sing-box/cmd/sing-box" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3" GO_BUILD_DIR="/home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3/.go_work/build" GO_BUILD_BIN_DIR="/home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3/.go_work/build/bin/linux_arm64" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/hzybb/immortalwrt/staging_dir/target-aarch64_generic_musl/usr/share/gocode/src" GOOS="linux" GOARCH="arm64" GO386="" GOAMD64="" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3=sing-box-1.9.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/include -I/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/fortify -I/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include" CGO_CXXFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3=sing-box-1.9.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/lib -L/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib -znow -zrelro" CGO_ENABLED=1 GOPATH="/home/hzybb/immortalwrt/build_dir/target-aarch64_generic_musl/sing-box-1.9.3/.go_work/build" GOCACHE="/home/hzybb/immortalwrt/tmp/go-build" GOMODCACHE="/home/hzybb/immortalwrt/dl/go-mod-cache" GOENV=off /usr/bin/env bash /home/hzybb/immortalwrt/feeds/packages/lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1678323167' -linkmode external -extldflags '-L/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/lib -L/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags " -X github.com/sagernet/sing-box/constant.Version=v1.9.3 -buildid '1678323167' -linkmode external -extldflags '-L/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/lib -L/home/hzybb/immortalwrt/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'" -tags "with_clash_api,with_dhcp,with_ech,with_gvisor,with_quic,with_utls,with_wireguard" 
Finding targets

Building targets
../../common/tls/reality_client.go:10:2: package crypto/ecdh is not in GOROOT (/home/hzybb/immortalwrt/staging_dir/hostpkg/lib/go-cross/src/crypto/ecdh)

[... cut ...]
    ERROR: package/feeds/small/sing-box failed to build.
make[2]: *** [package/Makefile:116: package/feeds/small/sing-box/compile] Error 1
make[2]: Leaving directory '/home/hbb/immortalwrt'
make[1]: *** [package/Makefile:110: /home/hbb/immortalwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/hbb/immortalwrt'
make: *** [/home/hbb/immortalwrt/include/toplevel.mk:230: world] Error 2

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

xlighting2017, 2024-06-13:
small/sing-box  是哪来的。。。

p-w-p, 2024-06-13:
https://github.com/kenzok8/small/issues