immortalwrt/immortalwrt #1335: 自编译R2s设备固件版本23.05.2，安装插件alist发生错误。

lysgwl, 2024-06-17:
### Describe the bug

自己编译的openwrt固件23.05.02版本，然而在安装luci-app-alist插件时，发生以下错误，不知道什么原因造成的，请各位大神指条明路：
root@wl-r2s:/tmp/packages_ci# opkg install luci-app-alist_1.0.13_all.ipk
Installing luci-app-alist (1.0.13) to root...
Configuring luci-app-alist.
//usr/lib/opkg/info/luci-app-alist.postinst: .: line 3: can't open '/etc/uci-defaults/50-luci-alist': No such file or directory

这种问题什么原因造成的，是编译的问题，还是插件的问题？还是缺少什么库造成的，如何修改呢？

### ImmortalWrt version

r27625-416c8c5c91

### ImmortalWrt release

23.05.2

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

r2s

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

xlighting2017, 2024-06-17:
你自己捣鼓出来一个 windows，然后从不知道哪里下载了一个 office安装文件， 装不上报错
找微软售后，不太合适吧...

lysgwl, 2024-06-17:
> Reference

那你知道吗？虚心请教哈。

lysgwl, 2024-06-17:
> 你自己捣鼓出来一个 windows，然后从不知道哪里下载了一个 office安装文件， 装不上报错 找微软售后，不太合适吧...

那你知道吗？虚心请教哈。

xlighting2017, 2024-06-17:
谁也不知道你自己怎么编的，编的时候配置了什么；

更不知道 luci-app-alist_1.0.13_all.ipk 哪里来的，里面是/有什么；

那怎么知道呢？
我猜编译的有问题，或者安装文件不对，肯定是这2个原因之一

lysgwl, 2024-06-17:
> 谁也不知道你自己怎么编的，编的时候配置了什么；
> 
> 更不知道 luci-app-alist_1.0.13_all.ipk 哪里来的，里面是/有什么；
> 
> 那怎么知道呢？ 我猜编译的有问题，或者安装文件不对，肯定是这2个原因之一

这个简单，luci-app-alist插件用的是sbwml下载的插件，配置信息没有什么特殊的，就是编译时候加载编译了几个插件。

xlighting2017, 2024-06-17:
还原使用原版 23.05.02 ,  任何需要的插件使用
```
opkg update
opkg install luci-app-xxxxxx <--具体软件名
```
即可

wolvth, 2024-06-17:
因为你的插件版本和23.05.02不匹配。sbwnl的release页面里并没有23.05的版本，要想使用他的版本得自己编译。
或者直接全都卸载掉，使用immortal自己的版本。按楼上的方法即可。

lysgwl, 2024-06-17:
> 因为你的插件版本和23.05.02不匹配。sbwnl的release页面里并没有23.05的版本，要想使用他的版本得自己编译。 或者直接全都卸载掉，使用immortal自己的版本。按楼上的方法即可。

额，明白了。版本问题这个可能性很大，但是之前编译过的版本，然后在安装sbwnl的release版本，ok就能通过，难道是个玄学问题？报错的问题，理解应该是个配置文件丢失。

closed 2024-06-17 by 1715173329