immortalwrt/immortalwrt #1353: BPI-R4 硬件流量分载无效

yuban10703, 2024-07-02:
### Describe the bug

防火墙的硬件流量分载好像是默认启用的,但是好像并没有生效(测速时占用CPU),ppe的bind里也没有连接
```
root@ImmortalWrt:~# cat /sys/kernel/debug/ppe*/bind
root@ImmortalWrt:~#
```

在openwrt的main分支上硬件流量分载是生效的,测速时无CPU占用,bind里也有连接

```
root@OpenWrt:~# cat /sys/kernel/debug/ppe*/bind
007e0 BND IPv4 5T orig=10.1.1.136:5653->101.95.52.102:80 new=222.67.134.24:5653->101.95.52.102:80 eth=0a:d5:17:47:e9:f4->d0:d0:4b:06:ec:cd etype=6488 vlan=0,0 ib1=205200ca ib2=00f09f0e packets=19640 bytes=1193136
007e4 BND IPv4 5T orig=10.1.1.136:5652->101.95.52.102:80 new=222.67.134.24:5652->101.95.52.102:80 eth=0a:d5:17:47:e9:f4->d0:d0:4b:06:ec:cd etype=6488 vlan=0,0 ib1=205200ca ib2=00f09f0e packets=20017 bytes=1212592
00fe8 BND IPv4 5T orig=10.1.1.136:5655->101.95.52.102:80 new=222.67.134.24:5655->101.95.52.102:80 eth=0a:d5:17:47:e9:f4->d0:d0:4b:06:ec:cd etype=6488 vlan=0,0 ib1=205200ca ib2=00f09f0e packets=10299 bytes=619202
00fec BND IPv4 5T orig=10.1.1.136:5654->101.95.52.102:80 new=222.67.134.24:5654->101.95.52.102:80 eth=0a:d5:17:47:e9:f4->d0:d0:4b:06:ec:cd etype=6488 vlan=0,0 ib1=205200ca ib2=00f09f0e packets=13238 bytes=795952
017d0 BND IPv4 5T orig=10.1.1.136:5657->101.95.52.102:80 new=222.67.134.24:5657->101.95.52.102:80 eth=0a:d5:17:47:e9:f4->d0:d0:4b:06:ec:cd etype=6488 vlan=0,0 ib1=205200ca ib2=00f09f0e packets=15570 bytes=939172
017d4 BND IPv4 5T orig=10.1.1.136:5656->101.95.52.102:80 new=222.67.134.24:5656->101.95.52.102:80 eth=0a:d5:17:47:e9:f4->d0:d0:4b:06:ec:cd etype=6488 vlan=0,0 ib1=205200ca ib2=00f09f0e packets=18515 bytes=1121690
00198 BND IPv4 5T orig=101.95.52.102:80->222.67.134.24:5655 new=101.95.52.102:80->10.1.1.136:5655 eth=0a:d5:17:47:e9:f3->8c:0e:60:67:9f:3a etype=0008 vlan=0,0 ib1=205000ca ib2=00f08501 packets=43381 bytes=65678834
0019c BND IPv4 5T orig=101.95.52.102:80->222.67.134.24:5654 new=101.95.52.102:80->10.1.1.136:5654 eth=0a:d5:17:47:e9:f3->8c:0e:60:67:9f:3a etype=0008 vlan=0,0 ib1=205000ca ib2=00f08501 packets=56421 bytes=85421394
00990 BND IPv4 5T orig=101.95.52.102:80->222.67.134.24:5653 new=101.95.52.102:80->10.1.1.136:5653 eth=0a:d5:17:47:e9:f3->8c:0e:60:67:9f:3a etype=0008 vlan=0,0 ib1=205000ca ib2=00f08501 packets=111697 bytes=169105162
00994 BND IPv4 5T orig=101.95.52.102:80->222.67.134.24:5652 new=101.95.52.102:80->10.1.1.136:5652 eth=0a:d5:17:47:e9:f3->8c:0e:60:67:9f:3a etype=0008 vlan=0,0 ib1=205000ca ib2=00f08501 packets=106872 bytes=161804208
019a0 BND IPv4 5T orig=101.95.52.102:80->222.67.134.24:5657 new=101.95.52.102:80->10.1.1.136:5657 eth=0a:d5:17:47:e9:f3->8c:0e:60:67:9f:3a etype=0008 vlan=0,0 ib1=205000ca ib2=00f08501 packets=80560 bytes=121967624
019a4 BND IPv4 5T orig=101.95.52.102:80->222.67.134.24:5656 new=101.95.52.102:80->10.1.1.136:5656 eth=0a:d5:17:47:e9:f3->8c:0e:60:67:9f:3a etype=0008 vlan=0,0 ib1=205000ca ib2=00f08501 packets=89388 bytes=135333432
root@OpenWrt:~#
```

### ImmortalWrt version

r0-dd4624e

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Self-built image

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r4=y
CONFIG_LUCI_LANG_zh_Hans=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_default-settings=y
CONFIG_PACKAGE_default-settings-chn=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-i18n-base-zh-cn=y
CONFIG_PACKAGE_luci-i18n-firewall-zh-cn=y
CONFIG_PACKAGE_luci-i18n-opkg-zh-cn=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
CONFIG_TARGET_ROOTFS_PARTSIZE=448
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-07-03:
可以试试移除 package/network/config/firewall4/patches/002-fix-adding-offloading-device.patch

yuban10703, 2024-07-03:
移除后工作正常,感谢
[image]

1715173329, 2024-07-03:
麻烦打印一下在带和不带此补丁下的 nft 规则，命令 `nft list ruleset`

yuban10703, 2024-07-03:
<details>
  <summary>
    不带此补丁的规则
  </summary>

```
root@ImmortalWrt:~# nft list ruleset
table inet fw4 {
        flowtable ft {
                hook ingress priority filter
                devices = { eth1, eth2, lan1, lan2, lan3 }
                flags offload
                counter
        }

        chain input {
                type filter hook input priority filter; policy drop;
                iif "lo" accept comment "!fw4: Accept traffic from loopback"
                ct state vmap { established : accept, related : accept } comment "!fw4: Handle inbound flows"
                tcp flags syn / fin,syn,rst,ack jump syn_flood comment "!fw4: Rate limit TCP syn packets"
                iifname "br-lan" jump input_lan comment "!fw4: Handle lan IPv4/IPv6 input traffic"
                iifname "pppoe-wan" jump input_wan comment "!fw4: Handle wan IPv4/IPv6 input traffic"
                jump handle_reject
        }

        chain forward {
                type filter hook forward priority filter; policy drop;
                meta l4proto { tcp, udp } flow add @ft
                ct state vmap { established : accept, related : accept } comment "!fw4: Handle forwarded flows"
                iifname "br-lan" jump forward_lan comment "!fw4: Handle lan IPv4/IPv6 forward traffic"
                iifname "pppoe-wan" jump forward_wan comment "!fw4: Handle wan IPv4/IPv6 forward traffic"
                jump handle_reject
        }

        chain output {
                type filter hook output priority filter; policy accept;
                oif "lo" accept comment "!fw4: Accept traffic towards loopback"
                ct state vmap { established : accept, related : accept } comment "!fw4: Handle outbound flows"
                oifname "br-lan" jump output_lan comment "!fw4: Handle lan IPv4/IPv6 output traffic"
                oifname "pppoe-wan" jump output_wan comment "!fw4: Handle wan IPv4/IPv6 output traffic"
        }

        chain prerouting {
                type filter hook prerouting priority filter; policy accept;
        }

        chain handle_reject {
                meta l4proto tcp reject with tcp reset comment "!fw4: Reject TCP traffic"
                reject comment "!fw4: Reject any other traffic"
        }

        chain syn_flood {
                limit rate 25/second burst 50 packets return comment "!fw4: Accept SYN packets below rate-limit"
                drop comment "!fw4: Drop excess packets"
        }

        chain input_lan {
                jump accept_from_lan
        }

        chain output_lan {
                jump accept_to_lan
        }

        chain forward_lan {
                jump accept_to_wan comment "!fw4: Accept lan to wan forwarding"
                jump accept_to_lan
        }

        chain accept_from_lan {
[... 251 lines cut ...]
                oifname "pppoe-wan" tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 egress MTU fixing"
        }

        chain mangle_input {
                type filter hook input priority mangle; policy accept;
        }

        chain mangle_output {
                type route hook output priority mangle; policy accept;
        }

        chain mangle_forward {
                type filter hook forward priority mangle; policy accept;
                iifname "pppoe-wan" tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 ingress MTU fixing"
        }
}
table inet dnsmasq {
        chain prerouting {
                type nat hook prerouting priority dstnat - 5; policy accept;
                meta nfproto { ipv4, ipv6 } udp dport 53 counter packets 9 bytes 605 redirect to :53 comment "DNSMASQ HIJACK"
        }
}
```

</details>

1715173329, 2024-07-03:
好的，非常感谢

1715173329, 2024-07-03:
暂时先 open，等找到修复办法再 close。

pomhg, 2024-08-06:
二者的区别在于是否device包含pppoe或者vlan，根据[https://www.kernel.org/doc/html/latest/networking/nf_flowtable.html](url)，flowtable只需要包含真实二层接口。
[image]

xlighting2017, 2024-08-06:
there were quite a lot discussion around flow-offloading, and few fixes are push to package `firewall4`

how about we bump the FW4 to the latest, and drop the 002-patch here?

ref:
[a long discussion of soft flow-offloading](https://github.com/openwrt/openwrt/issues/10224)
https://github.com/openwrt/openwrt/issues/13410

https://github.com/openwrt/firewall4/commit/e00958884416f59b273595f941d198de63acc1dd
https://github.com/openwrt/firewall4/commit/dfbcc1cd127c78fc61bb870d36d2512b571d223b

closed 2024-08-08 by 1715173329

brada4, 2024-09-02:
@xlighting2017 there are multiple problems with offloading:
1/ oversized packets knock out offloaded flow back to conntrack slowpath (essentially accept established/related). most notable with wg, pppoe ie interfaces with !=1500 mtu. Fixed here: https://github.com/openwrt/firewall4/commit/698a53354fd280aae097efe08803c0c9a10c14c2
2/ no need to add higher level devices BUT then output of the flow-state is fixated on a low level device https://www.kernel.org/doc/html/latest/networking/nf_flowtable.html#limitations
2 patches referred get around them by setting offload endpoints to devices where firewall actually forwards, keeping ARP and free movement of MAC between physical ports.
2a/ https://github.com/openwrt/openwrt/issues/16184 skips buggy drivers by accident
2b/ accidentally docker0 got free soft offload
3/ to be continued.