immortalwrt/immortalwrt #1379: hi,tianling. re-cp-03 use 23.05.2 and 23.05.3 some times reboot router , the wan  all can‘t get online

aYangQT, 2024-07-22:
### Describe the bug

hi,tianling. re-cp-03 use 23.05.2 and 23.05.3 some times reboot router , the wan  all can‘t get online

### ImmortalWrt version

 23.05.3 (r27917-81a1f98d5b) and  23.05.2 (r27625-416c8c5c91)

### ImmortalWrt release

 23.05.3 (r27917-81a1f98d5b)

### ImmortalWrt target/subtarget

 23.05.3 (r27917-81a1f98d5b)

### Device

re-cp-03

### Image kind

Official downloaded image

### Steps to reproduce

刷gpt
dd if=immortalwrt-23.05.2-mediatek-filogic-jdcloud_re-cp-03-gpt.bin of=/dev/mmcblk0 bs=512 seek=0 count=34 conv=fsync
刷bl2
echo 0 > /sys/block/mmcblk0boot0/force_ro
dd if=/dev/zero of=/dev/mmcblk0boot0 bs=512 count=8192 conv=fsync
dd if=immortalwrt-23.05.2-mediatek-filogic-jdcloud_re-cp-03-preloader.bin of=/dev/mmcblk0boot0 bs=512 conv=fsync
刷fip
dd if=/dev/zero of=/dev/mmcblk0 bs=512 seek=13312 count=8192 conv=fsync
dd if=immortalwrt-23.05.2-mediatek-filogic-jdcloud_re-cp-03-bl31-uboot.fip of=/dev/mmcblk0 bs=512 seek=13312 conv=fsync

The same way to brush the official openwrt 23.05.4 and immortalwrt 23.05.2 and 23.05.3 have the problem that the wan port cannot obtain the ip after restarting
Immortalwrt 23.05.2 sometimes modify wan mac address can be normal Internet access, but after restarting the offer can not go online.

同样方法刷官方openwrt 23.05.4和immortalwrt 23.05.2 和23.05.3都有重启后wan口不能获取到ip这个问题
immortalwrt 23.05.2 有时候修改wan口mac地址可以正常上网，但重启后优惠不能上网

### Actual behaviour

sometimes reboot router

### Expected behaviour

the wan port cannot obtain the ip after restarting

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-07-22:
No log = no solution.

aYangQT, 2024-07-22:
> 没有日志 = 没有解决方案。

下班后我重刷immortalwrt下载log，昨天已经刷成加菲猫的了

aYangQT, 2024-07-29:
大佬可以分享下re-cp-03设备的官方imm刷机方法吗，然后我这边对比下刷机步骤有没有问题

aYangQT, 2024-07-29:
我发现有的人刷机后用拨号模式就是正常的，我的都是dhcp获取的上网ip，然后有没有交流群或者联系大佬的方式，我这边复现后联系大佬，最近在忙

libiunc, 2024-07-29:
> 大佬可以分享下re-cp-03设备的官方imm刷机方法吗，然后我这边对比下刷机步骤有没有问题

善用搜索
https://github.com/search?q=repo%3Aimmortalwrt%2Fimmortalwrt+re-cp-03&type=commits

https://github.com/immortalwrt/immortalwrt/commit/c0c3234e17207a9287a08757fc1752490144a1cd

aYangQT, 2024-07-29:
> > 大佬可以分享下re-cp-03设备的官方imm刷机方法吗，然后我这边对比下刷机步骤有没有问题
> 
> 善用搜索 https://github.com/search?q=repo%3Aimmortalwrt%2Fimmortalwrt+re-cp-03&type=commits
> 
> [货号：C0C3234](https://github.com/immortalwrt/immortalwrt/commit/c0c3234e17207a9287a08757fc1752490144a1cd)

感谢解答，和我刷机的步骤完全一致，等有时间，我再复现下，把log发出来

aYangQT, 2024-08-04:
> 无日志 = 无解决方案。

你好，天灵，我这边将日志下载下来了，怎么传输给您，可以通过邮箱或者私信发您吗

1715173329, 2024-08-04:
发在这里。

aYangQT, 2024-08-04:
> 发在这里。

好的，感谢回复，log压缩密码：tianling
[re-cp-03_log.zip](https://github.com/user-attachments/files/16486627/re-cp-03_log.zip)

1715173329, 2024-08-04:
先换网线

aYangQT, 2024-08-04:
> 先换网线

更换了多条网线，wan口连接上级路由2.5G口web主页识别为2.5G，1G识别为100兆，均不能获取到ip

1715173329, 2024-08-04:
```patch
--- a/target/linux/mediatek/dts/mt7986a-jdcloud-re-cp-03.dts
+++ b/target/linux/mediatek/dts/mt7986a-jdcloud-re-cp-03.dts
@@ -133,8 +133,8 @@
 		reg = <6>;
 
 		reset-gpios = <&pio 6 GPIO_ACTIVE_LOW>;
-		reset-assert-us = <10000>;
-		reset-deassert-us = <50000>;
+		reset-assert-us = <15000>;
+		reset-deassert-us = <68000>;
 		realtek,aldps-enable;
 	};
 
```

patch 下编译试试。

aYangQT, 2024-08-04:
好的，这是多次重启后wan口恢复正常的日志，log压缩密码：tianling
[多次重启后wan口恢复正常的日志.zip](https://github.com/user-attachments/files/16486778/wan.zip)

aYangQT, 2024-08-04:
我之前编译re-cp-03的op源码好像没有界面，最近有使用加菲猫的云编译过H大的源码，imm官方是否有类似的云编译方法，或者等大佬更新，我再测试也可以

aYangQT, 2024-08-05:
昨晚改成pppoe拨号，还是imm官方的23.5.03稳定版，也是会出现dhcp一样的wan口不能获取ip，1G速率，web页面显示协商成100M。重启几次目前正常了。再次重启也会复现。有时间研究下怎么编译大佬给的修改后的代码

aYangQT, 2024-08-06:
[immortalwrt-mediatek-filogic-jdcloud_re-cp-03.zip](https://github.com/user-attachments/files/16504154/immortalwrt-mediatek-filogic-jdcloud_re-cp-03.zip)
编译环境wsl2 ubuntu24 ,按照大佬修改后的代码选择re-cp-03,然后root分区设置2G空间，默认好像160M太小了，其他没有修改。晚上回去更新下sysupgrade.itb测试wan口是不是正常，这种方法之前编译op好像没有luci界面，不知道imm是不是正常

aYangQT, 2024-08-06:
> ```diff
> --- a/target/linux/mediatek/dts/mt7986a-jdcloud-re-cp-03.dts
> +++ b/target/linux/mediatek/dts/mt7986a-jdcloud-re-cp-03.dts
> @@ -133,8 +133,8 @@
>  		reg = <6>;
>  
>  		reset-gpios = <&pio 6 GPIO_ACTIVE_LOW>;
> -		reset-assert-us = <10000>;
> -		reset-deassert-us = <50000>;
> +		reset-assert-us = <15000>;
> +		reset-deassert-us = <68000>;
>  		realtek,aldps-enable;
>  	};
>  
> ```
> 
> patch 下编译试试。

拉取今天最新的23.05.3版本分支，按照大佬说的修改超时后，选择re-cp-03配置，设置root分区大小从160M改成2G(增加软件包安装空间)，通过web更新sysupgrade.itb，经过4次网页重启和1次断电重启后，wan口均正常pppoe获取ip，后续再测试是否有其他问题，感谢大佬开发并提供无私支持

aYangQT, 2024-08-06:
不过主页显示的固件版本和imm官方下载的23.05.3是完全一样的，但是我看日志有我的编译环境的名称，应该是成功更新了，不过日志日期好像锁定在8月5号问题不大应该

aYangQT, 2024-08-07:
刚刚拉取并编译了一版imm,改了大佬的超时代码。看了下日志，好像有下面的日志报错，不过使用正常，这些报错应该问题不大吧
dropbear SSH服务启动时出现了 Failed listening on '22': Error listening: Address not available 错误
日志中多次出现 No default route present, setting ra_lifetime to zero!
无线设备 mt798x-wmac 初始化时出现了 eeprom load fail 错误，好像使用了默认的固件。

Tue Aug  6 19:31:38 2024 authpriv.warn dropbear[2630]: Failed listening on '22': Error listening: Address not available
Tue Aug  6 19:31:38 2024 authpriv.info dropbear[2630]: Not backgrounding

Tue Aug  6 19:31:41 2024 daemon.err odhcp6c[2873]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Tue Aug  6 19:31:42 2024 daemon.warn odhcpd[1822]: No default route present, setting ra_lifetime to zero!
Tue Aug  6 19:31:45 2024 daemon.warn odhcpd[1822]: No default route present, setting ra_lifetime to zero!

[    5.429670] random: 24 urandom warning(s) missed due to ratelimiting
[    5.880397] mt798x-wmac 18000000.wifi: HW/SW Version: 0x8a108a10, Build Time: 20221012174648a
[    5.880397] 
[    5.986787] mt798x-wmac 18000000.wifi: WM Firmware Version: ____000000, Build Time: 20221012174725
[    6.055896] mt798x-wmac 18000000.wifi: WA Firmware Version: DEV_000000, Build Time: 20221012174937
[    6.202536] mt798x-wmac 18000000.wifi: eeprom load fail, use default bin
[    6.210886] mt798x-wmac 18000000.wifi: registering led 'mt76-phy0'
[    6.218942] mt798x-wmac 18000000.wifi: registering led 'mt76-phy1'

closed 2024-08-09 by 1715173329